Atmosphere-NX/Atmosphere #11: Create a framework for generically overriding title contents.

SciresM, 2018-02-17:
We'll want to be able to replace title executables/RomFS contents.

Should this work the way Luma does it?

Should we go for custom BKTRs (official RomFS patch format), or a LayeredFS style solution for RomFS, in the long term?

DioVento, 2018-02-17:
As someone who makes and distributes 3DS ROM hacks, LayeredFS is a godsend: I can distribute "drag and drop" packages that don't require my (often very knowledge-limited) user base to compile things. If I were releasing things like modified update CIAs, I'd be restricted by the anti-piracy rules on the 3DS hacking reddits and GBATemp.

SciresM, 2018-02-17:
@DioVento Yeah, I understand well the benefits of LayeredFS. The counter-argument is, of course, that BKTR is designed specifically to allow official devs to do LayeredFS style things -- BKTR is a format designed around merging two separate archives into a bigger virtual archive (official content uses it for adding new files to RomFS). 

Effectively, you'd compile an archive containing your new files and the new RomFS layout (we'd write a tool if we went down this path to make BKTRs), and then place it on the SD. It would behave like LayeredFS as far as the app is concerned, and you'd only be including the content you changed in the compiled BKTR archive.

Of course, this has downsides, too -- it's more complicated from an end user perspective than LayeredFS. Needs more thought :)

CuriousTommy, 2018-02-18:
If the tool to manipulate BKTR files is an easy-to-use cross-platform GUI application, then I think using a custom BKTR should be fine. 

I mean, why reinvent the wheel when you can official usable functions from Nintendo. It provides most of the benefits that that people expect with LayeredFS with the only downside of it all being stored in a file.

Niramo, 2018-02-18:
Doesn't BKTR interfere with game updates?

HoLLy-HaCKeR, 2018-02-18:
Would this mean only "borrowing" the format and surrounding functions, or also installing romhacks like updates?

SciresM, 2018-08-02:
This is completed.

closed 2018-08-02 by SciresM