Atmosphere-NX/Atmosphere #12: Decide on a moderate to long-term custom sysmodule plan.

SciresM, 2018-02-17:
It'll be a while before this is urgent, but we should figure out what sysmodules we'd like to potentially re-implement/what kinds of custom services/API we'd like for our cfw to provide to the user and user homebrew.

TuxSH, 2018-02-17:
I think it should be (3.0+ ?) loader, sm and boot at first.

A custom sysmodule seems like a nice idea as well.

SciresM, 2018-02-17:
Yeah, I think I agree with that. Those are all _very_ light.

From an extensibility perspective, I also think we might want to tackle NCM in the moderate term. I'd be happy to take that one on, I'm very familiar with how NCM works on the inside.

MRiCEQB, 2018-02-17:
If you don't reimplement golf, you can basically stop the entire project imho.

Swiftloke, 2018-02-17:
I understand you were memeing, but nonetheless, that would be distribution of copyrighted content (the Golf ROM) and thus highly illegal.

asiekierka, 2018-02-21:
@Swiftloke Nobody said they have to redistribute the golf ROM; a custom implementation of a golf game would probably be sufficient.

SciresM, 2018-04-22:
Targeted for initial release:
sm, loader, boot. boot2?

Targeted for after initial release:
boot2? SPL, PM, NCM.

Custom sysmodules that may be desirable:
mitm (framework for man-in-the-middle attacking other services)
layeredfs (RomFS framework, based around mitm)

closed 2018-04-22 by SciresM