Atmosphere-NX/Atmosphere #13: Get a framework in place for launching homebrew from the home menu.

SciresM, 2018-02-17:
This may take the form of our patching qlaunch, or our replacing qlaunch's layout archives. Who knows? We'll have to figure out how we want to handle this, but it's a top priority item to support.

iGPLTV, 2018-04-24:
the words "home menu" and "launching homebrew" make me happy.

thomasnet-mc, 2018-07-02:
Wouldn't we just go with installing an application? I know Rei is working on that with BogInstaller. We do need an NCA generator though.

PyroFlareX, 2018-07-08:
in my head, what would make sense would be for CFW to not require security install tickets on carts and installable, so besides making romsbuilder be able to make nca's, it'd work.

(this might not even reasonable)

fincs, 2018-09-20:
Running homebrew shouldn't involve installing custom titles. Installing custom titles is risky because it entails a ban from Nintendo's services.

ThatNerdyPikachu, 2018-09-20:
Oh, right. Forgot about that...

closed 2018-10-17 by SciresM