Atmosphere-NX/Atmosphere #3: Implement a generic patching system for Kernel/FIRM sysmodules.

SciresM, 2018-02-17:
Decide on a standardized format, and implement in the stage 1 loader. Should be flexible enough to pattern match -- see luma-style stuff. 

Is this best done via a custom "loader" system module? Needs design considerations.

Valeri0p, 2018-02-19:
Sorry to bother you, probably you already know about [this approach](https://yifan.lu/2016/11/17/designing-a-cfw-framework), but just in case you need some inspiration 😉
Thx for all your work!

Resaec, 2018-09-22:
It is also possible to make the mask binary (`uint64_t` or (typed) `uint128_t`, if the pattern gets that long) to save space (1bit <> 8bit) and use fast binary operations to check for wildcards.
```if(mask & 1 << loop_i)```

Selver-gba, 2018-10-12:
This is a difficult question to answer, without being clear on the types of patches that require support.
At a first level, I second @Valeri0p's suggestion to review [yifan lu's CFW framework](https://yifan.lu/2016/11/17/designing-a-cfw-framework).  For any patching that can be done at the exported API layer, this seems a good starting point.

The pattern matching concept is also likely going to be needed.  

The hard part (imho) is making multiple patches that affect the same area work.  Even with the above framework, which hook goes "first" is a very hard problem.

A similar problem appears to have been (attempted?) solved for [file system filter driver ordering](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/load-order-groups-for-file-system-filter-drivers) in Windows.  At a high level, they define multiple types of filter drivers, with a pre-defined order that they get applied.

Perhaps something similar is appropriate for a patch framework?

Resaec, 2018-10-17:
Is pattern matching still desired?
I finished coding yesterday, but didn't push or PR it as I thought I needed to include a working patching system at the same time. I will push it to my fork in a distinct branch if anyone wants to take a look.

As kind of a benchmark
Debian (gcc) - this seems to be cached, as there is no way for matching on a 1GB file being this fast
```
nxdev@nxdev:~/memtest$ ./test
Malloc took 0.000010s for 1048.576000 MB
File read to memory took 0.561273s to complete
memSearchASLR took 1.856251s and location is 0x3B9ACA00
memSearch took 0.684708s and location is 0x3B9ACA00
```
Windows (MSVC)
```
Malloc took 0.475000s for 1048.576000 MB
File read to memory took 1.475000s to complete
memSearchASLR took 8.280000s and location is 0x3B9ACA00
```