Atmosphere-NX/Atmosphere #9: Create an emunand management tool.

SciresM, 2018-02-17:
We'll need a tool to back up a sysnand to emunand, and to format an SD card to contain the dumped NAND sectors.

Generally, this should be a NAND dump/restoration tool.

Zero-machine, 2018-07-10:
You can use a Magic NS Adapter in the meantime, it supports all the main bluetooth controllers and can be used both in the docked and tablet mode, although this last one needs also a cable connected to the dongle. [https://www.amazon.com/Mayflash-MAGIC-NS-Wireless-Controller-Fighting-PC/dp/B074JZPS39#customerReviews](url)

Huskatron, 2018-12-18:
Hurry up.

deadair3210, 2018-12-18:
> Hurry up.

Because rushing someone doing work for free always ends well.....

Huskatron, 2019-01-02:
😭

Brawl345, 2019-01-02:
Can you stop spamming this issue? It's done when it's done.

SciresM, 2019-12-30:
"Hekate"

closed 2019-12-30 by SciresM