Atmosphere-NX/Atmosphere #59: Debug process?

afturner, 2018-03-14:
Hi there! I'm a dev with zero console hacking experience (aside from following the 3DS guide lol). 

I'd like to work on Atmosphere but would like some information on how exactly you debug/run. You compile... and then what? Put it on an SD.. and? I'm probably just missing the entrypoint guide but whatever. 

Thanks!

916253, 2018-03-15:
There's not currently a public entrypoint for running this, so contribution is a bit difficult from the general public perspective-- public contribution will become easier in the coming months once some of the currently private exploits are made public.

mariogamer2, 2018-03-15:
Actually, it is possible *if* you can implement jamais vu on a 1.0.0 switch (I think).

jason0597, 2018-04-24:
well now with TegraRCM released things have changed

112madgamer, 2018-04-26:
I have it compiled just unsure how to run it

hexkyz, 2018-06-04:
Closing this issue since the project is not ready for user support yet.

closed 2018-06-04 by hexkyz