Atmosphere-NX/Atmosphere #74: Unable to build Atmosphère : unrecognized command line option '-mthumb'

matteyeux, 2018-05-01:
Hey there, I'm trying to build Atmosphère on Debian GNU/Linux 8 (jessie), but I got these errors :
```bash
$ make DEVKITPRO=/opt/devkitpro/ DEVKITARM=/opt/devkitpro/devkitA64
make -C exosphere 
make[1]: Entering directory '/home/mathieu/Documents/Atmosphere/exosphere'
make[2]: Entering directory '/home/mathieu/Documents/Atmosphere/exosphere/bpmpfw'
aarch64-none-elf-gcc -march=armv4t -mtune=arm7tdmi -mthumb -mthumb-interwork -g -O2 -ffunction-sections -fdata-sections -fomit-frame-pointer -fno-inline -std=gnu11 -Werror -Wall   -c -o build/emc.o src/emc.c
aarch64-none-elf-gcc: error: unrecognized command line option '-mthumb'
aarch64-none-elf-gcc: error: unrecognized command line option '-mthumb-interwork'
Makefile:56: recipe for target 'build/emc.o' failed
make[2]: *** [build/emc.o] Error 1
make[2]: Leaving directory '/home/mathieu/Documents/Atmosphere/exosphere/bpmpfw'
Makefile:61: recipe for target 'bpmpfw/out/bpmpfw.bin' failed
make[1]: *** [bpmpfw/out/bpmpfw.bin] Error 2
make[1]: Leaving directory '/home/mathieu/Documents/Atmosphere/exosphere'
Makefile:8: recipe for target 'exosphere' failed
make: *** [exosphere] Error 2
```
Am I doing something wrong ?

fincs, 2018-05-01:
DEVKITARM is supposed to point to devkitARM, not devkitA64.

matteyeux, 2018-05-01:
Fixed ! 
Thank you for your help

closed 2018-05-01 by matteyeux

chengstone, 2018-05-18:
hi, when I do build Atmosphere at Ubuntu , got following errors, I Know its a uul type problem, but I dont know how to fix that:

`make -C exosphere 
make[1]: Entering directory '/home/chengstone/Downloads/Atmosphere/exosphere'
arm-none-eabi-gcc -march=armv8-a -mtune=cortex-a57 -g -O2 -ffunction-sections -fdata-sections -fomit-frame-pointer -std=gnu11 -Werror -Wall   -c -o build/sealedkeys.o src/sealedkeys.c
arm-none-eabi-gcc -march=armv8-a -mtune=cortex-a57 -g -O2 -ffunction-sections -fdata-sections -fomit-frame-pointer -std=gnu11 -Werror -Wall   -c -o build/coldboot_main.o src/coldboot_main.c
In file included from src/coldboot_main.c:4:0:
src/coldboot_main.c: In function 'coldboot_main':
src/memory_map.h:132:53: error: large integer implicitly truncated to unsigned type [-Werror=overflow]
 #define TZRAM_GET_SEGMENT_ADDRESS(segment_id)       (TUPLE_FOLD_LEFT_2(EVAL(segment_id), _MMAPTZS, PLUS) EVAL(TZRAM_SEGMENT_BASE))
                                                     ^
src/coldboot_main.c:12:24: note: in expansion of macro 'TZRAM_GET_SEGMENT_ADDRESS'
     uintptr_t pk2ldr = TZRAM_GET_SEGMENT_ADDRESS(TZRAM_SEGMENT_ID_PK2LDR);
                        ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:71: recipe for target 'build/coldboot_main.o' failed
make[1]: *** [build/coldboot_main.o] Error 1
make[1]: Leaving directory '/home/chengstone/Downloads/Atmosphere/exosphere'
Makefile:8: recipe for target 'exosphere' failed
make: *** [exosphere] Error 2
`

is there has some compile arguments? like -m64? or is this a c13, c15, c17 features?
I only know use unsigned long long xxx replace #define xxx, but the work is huge.
so you guys how to build that correctly? pls help.

intraum, 2018-06-24:
i'm having the same problem, even when i point DEVKITARM to devkitARM