Atmosphere-NX/Atmosphere #82: Multi Language support

Philipp97714, 2018-05-05:
I would like to constribute something but don't know what. If you want to I can translate English strings to German for multi language support.

Sonixier, 2018-05-05:
I speak german too.

duracell, 2018-05-05:
There are many german speakers around :D

MrMaleficus, 2018-05-05:
I can translate the English strings to French 😊

slock83, 2018-05-05:
I can also work on the French translation (translating and/or checking)

Strat00s, 2018-05-05:
Hell yea. I can translate strings to the most useful language ever. Czech :D

YayIguess, 2018-05-05:
Probably know enough Russian to translate a fair bit, know some people good with Afrikaans and some other languages as well.

thomleg50, 2018-05-05:
I can translate to french

Philipp97714, 2018-05-05:
Can someone, who is working on the code, tell us, if a translation is possible/helpful?

flopglop, 2018-05-05:
I could help to translate Atmosphere into Spanish

Kleei, 2018-05-05:
i can translate to russian / romanian

Suinkai, 2018-05-05:
I can help translating to Spanish too

FearlessTobi, 2018-05-05:
I can help with all the German translation stuff too.

Philipp97714, 2018-05-05:
Please don't offer any translations before we know if they are helpful

CreedVI, 2018-05-05:
I can translate to German and Esperanto!

frankthedead, 2018-05-06:
I can translate to Portuguese.

SirNapkin1334, 2018-05-06:
I can do minor German translations.

deWasbeer, 2018-05-09:
I can help with dutch!

Deathwink, 2018-05-09:
I can translate to Italian

hexkyz, 2018-05-09:
Appreciate the enthusiasm, but Atmosphère currently has no need for translations. If/when this changes, we will announce it.

closed 2018-05-09 by hexkyz