Atmosphere-NX/Atmosphere #96: Readme file change suggestion

PyroFlareX, 2018-05-13:
For people like me who want to see and learn how this is done, but are annoyed and confused about how projects like these are structured, I say this:

Please include a readme section (or another text file) in how this project is structured rather than people like me wasting time reading C/C++ code and reading up again on switch stuff to find out what each does.

All in just an idea, and thank you for reading this as well as making this.

encounter, 2018-05-13:
>rather than people like me wasting time reading C/C++ code and reading up again on switch stuff

Both of those things are necessary to contribute to the project, and would not be a waste of time.
Unfortunately there's not much here that doesn't require a deep understanding of Switch internals.

Sorixelle, 2018-05-13:
If you're still interested in contributing, the [Projects](https://github.com/Atmosphere-NX/Atmosphere/projects) tab has a brief overview of the components.

PyroFlareX, 2018-05-13:
k thanks then! Thank you again for your hard work at this!

huangya90, 2018-05-16:
Yes. Looking forward to more documents besides https://github.com/Atmosphere-NX/Atmosphere/projects

hexkyz, 2018-06-04:
The current Projects tab provides an amount of information proportional to the current project's state. After an usable build is released, better documentation will follow.

closed 2018-06-04 by hexkyz