Atmosphere-NX/Atmosphere #101: Improve end-users communication on the projet

Arno500, 2018-05-14:
Hi !

I recently tried to run homebrews on the Switch with @ktemkin's exploit. In fact, it was quite diffucult since nothing is really clear and explained (to the end user I mean) and the sort-of wiki/guide to Switch dev was hard to found.

I have the competences needed to make a website (maybe hosted on GH pages ?) and am currently following a formation for this (also multimedia, etc...) + I'm willing to drink coffee with this.

This would be awesome to group all tutorials in "one" on THIS official website for the project, with all the steps needed to boot the payload and run the CFW.
I can use the assets already made for the project (from @jaames) too.

DevDivega, 2018-05-14:
Unfortunately this CFW has not been released for end users yet. It is currently in development and not meant to be used for any purpose other than development. Hang in there, the devs are making a lot of progress and once it is released I am sure Plialect will release a guide on setting it up!

natinusala, 2018-05-14:
There's no end-users communication because there are no end users just yet. Give it some time.

Arno500, 2018-05-14:
It's still lacking a front page, yet ;)

natehaxx, 2018-05-14:
Still its not made for End Users... people should calm down until its time

Arno500, 2018-05-14:
Then why it has assets ? A plain'ol text readme should be enough then...
I'm not complaining, it may be better to be full-public, or only dev oriented and keep everything "hidden"...
I'm closing the issue for now, so

closed 2018-05-14 by Arno500