Atmosphere-NX/Atmosphere #99: Underclocking

cualquiercosa327, 2018-05-14:
Hi.I would suggest if the cpu/gpu frecuences can be underclocked (an example i think than games as puyo puyo tennis doesnt need full frecuency speed for running and this would make the batery give extra time) as we can do on psp.
Thanks

cualquiercosa327, 2018-05-14:
i found than it can be set to 200 mhz.I think than it would be perfect for some homebrews and emulators than doesnt need so much frecuency.(As example nds and psp has a lot of emulators and most run on this (psp case,also 333 mhz) or lower frecuency )

adamnock, 2018-05-14:
I would suspect there is already a battery management system running which takes care of the majority of what this would achieve. (maybe not, I dont know).

Still, if there's any difficulty in having Homebrew utilise such a system, it might have benefit.

hexkyz, 2018-05-14:
Invalid issue

closed 2018-05-14 by hexkyz