Atmosphere-NX/Atmosphere #111: build error

thomleg50, 2018-05-21:
when i compile the source i'm unable to compile stratoshere and fusee secondary because of error : 

`tompi@DESKTOP-UG28LC8 MSYS ~/onedrive/documents/atmosphere-master
$ make
make -C fusee all
make[1] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/fusee »
make -C fusee-primary all
make[2] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/fusee/fusee-primary »
make[3]: rien à faire pour « all ».
make[2] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/fusee/fusee-primary »
make -C fusee-secondary all
make[2] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/fusee/fusee-secondary »
make[3] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/exosphere »
make[4] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/exosphere/bpmpfw »
make[5]: rien à faire pour « all ».
make[4] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/exosphere/bpmpfw »
make[4]: rien à faire pour « all ».
make[3] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/exosphere »
make[3] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/thermosphere »
make[4]: rien à faire pour « all ».
make[3] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/thermosphere »
make[3] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/stratosphere »
make -C libstratosphere all
make[4] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/stratosphere/libstratosphere »
make[5]: « /home/tompi/onedrive/documents/atmosphere-master/stratosphere/libstratosphere/lib/libstratosphere.a » est à jour.
make[5]: « /home/tompi/onedrive/documents/atmosphere-master/stratosphere/libstratosphere/lib/libstratosphered.a » est à jour.
make[4] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/stratosphere/libstratosphere »
make -C loader all
make[4] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/stratosphere/loader »
make[5]: rien à faire pour « all ».
make[4] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/stratosphere/loader »
make -C pm all
make[4] : on entre dans le répertoire « /home/tompi/onedrive/documents/atmosphere-master/stratosphere/pm »
pm_main.cpp
aarch64-none-elf-g++ -MMD -MP -MF /home/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/build/pm_main.d -g -Wall -O2 -ffunction-sections -march=armv8-a -mtune=cortex-a57 -mtp=soft -fPIE -DDISABLE_IPC -I/home/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/include -I/opt/devkitpro/portlibs/switch/include -I/opt/devkitpro/libnx/include -I/home/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/../libstratosphere/include -I/home/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/build -D__SWITCH__ -DSM_ENABLE_SMHAX -fno-rtti -fno-exceptions -std=gnu++17 -c /home/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/source/pm_main.cpp -o pm_main.o
C:/Users/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/source/pm_main.cpp: In function 'void __appInit()':
C:/Users/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/source/pm_main.cpp:71:10: error: 'smManagerInitialize' was not declared in this scope
     rc = smManagerInitialize();
          ^~~~~~~~~~~~~~~~~~~
C:/Users/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/source/pm_main.cpp:71:10: note: suggested alternative: 'splManuInitialize'
     rc = smManagerInitialize();
          ^~~~~~~~~~~~~~~~~~~
          splManuInitialize
C:/Users/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/source/pm_main.cpp: In function 'void __appExit()':
C:/Users/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/source/pm_main.cpp:93:5: error: 'smManagerExit' was not declared in this scope
     smManagerExit();
     ^~~~~~~~~~~~~
C:/Users/tompi/onedrive/documents/atmosphere-master/stratosphere/pm/source/pm_main.cpp:93:5: note: suggested alternative: 'splManuExit'
     smManagerExit();
     ^~~~~~~~~~~~~
     splManuExit
make[5]: *** [/opt/devkitpro/devkitA64/base_rules:14: pm_main.o] Error 1
make[4]: *** [Makefile:118: build] Error 2
make[4] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/stratosphere/pm »
make[3]: *** [Makefile:12: pm] Error 2
make[3] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/stratosphere »
make[2]: *** [Makefile:123: check_stratosphere] Error 2
make[2] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/fusee/fusee-secondary »
make[1]: *** [Makefile:8: fusee-secondary] Error 2
make[1] : on quitte le répertoire « /home/tompi/onedrive/documents/atmosphere-master/fusee »
make: *** [Makefile:5: fusee] Error 2

tompi@DESKTOP-UG28LC8 MSYS ~/onedrive/documents/atmosphere-master
`

natehaxx, 2018-05-21:
its not even ready for the public , try to update to the latest DevkitPro  but again this is not ready for Endusers to Play with.

thomleg50, 2018-05-21:
where the latest devkit ?

thomleg50, 2018-05-21:
i updated but i've got this error : https://pastebin.com/Du5xCiuf

TuxSH, 2018-05-21:
Update switch-tools too.

Closing.

closed 2018-05-21 by TuxSH

thomleg50, 2018-05-21:
ok thanks