Atmosphere-NX/Atmosphere #112: Idea Add ability to disable Nintendo network connectivity when atmosphere is running

fennectech, 2018-05-23:
I suggest adding the ability to turn off Nintendo network connectivity while atmosphere runs to reduce chances of a ban.

SciresM, 2018-05-24:
This is a native feature, just enable airplane mode.

closed 2018-05-24 by SciresM

drofdoom, 2018-05-24:
Disable Nintendo network by blocking them from your network via router or firewall.

fennectech, 2018-05-24:
That works.  But its not perfect. Ill look into it

sonofskywalker3, 2018-05-24:
That only works if you want it permanent, and it's a pain to have to change it every time you boot back to OFW. I agree an option to block Nintendo network traffic on CFW would be a great future addition.

fennectech, 2018-05-24:
He gets it!!

fennectech, 2018-05-24:
Thanks for all the hard work on atmosphere.  Its going to be GREAT and you shal become our homebrew god!

drofdoom, 2018-05-24:
But it's emunand ?

fennectech, 2018-05-24:
Nah. We’re talking about users that want to not use Nintendo network only when using CFW   And boot into OFW to play online with a reduced risk of being banned

drofdoom, 2018-05-24:
Use a proxy block the addresses with the port for your proxy added at the end for cfw and on ofw leave it open .

fennectech, 2018-05-24:
That could work.  Using a second copy of your WiFi profile with no proxy and the main auto connect profile with a proxy.    But its still not perfect and just a setting obeyed when atmosphere launches by default.