Atmosphere-NX/Atmosphere #113: Looking for list of compile dependencies

fennectech, 2018-05-25:
From reading the other issues I’ve infired I need libnx (compiling now) and switch-tools (unsure where to obtain.)   Are there any others I need and can you point me to their locations.  Thank you in advance.     Sorry if I’m waiting your time.

drofdoom, 2018-05-25:
Google switch tools switch brew

Sorixelle, 2018-05-25:
If memory serves me correctly, all you should need is devkitARM and devkitA64. They should have all the toolchains and libraries that you need.
‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

Filo97, 2018-05-25:
You just need a clean devkitpro install. I installed on windowsv with all consoles selected and was able to compile just fine. Have to remember, however, to define I_KNOW_WHAT_I_AM_DOING in fusee primary.

fennectech, 2018-05-25:
i got everything going   but i define I_KNOW_WHAT_I_AM_DOING in main.c right?

Traiver, 2018-05-25:
Even if you do that, I‘m still unsure if you know what you’re doing...
-- 
Von Patricks iPad gesendet

SciresM, 2018-05-25:
If you need to ask whether you know what you're doing, please consider whether you may not know what you're doing.

fennectech, 2018-05-25:
HAHA  i have no idea what im doing but im learning.   But ive gotten stratosphere compiled   and working on thermosphere and fusee (wich is the RCM payload correct)      Ive got a nand backup incase things get really bad

closed 2018-05-25 by SciresM

fennectech, 2018-05-25:
they dont all compile at one make from the root of the repo  and thermosphere and fusee dont compile for me  so im reading through the output to figure out whats holding it up.    Ill keep an eye on this though   It looks like it could be alot of fun.    Thanks :D