Atmosphere-NX/Atmosphere #115: i've got error

thomleg50, 2018-06-01:
in fusee secondary i've got this error : Failed to mount at least one partition: I/O error

if this sd related can you respond me ?

crc-32, 2018-06-01:
It is SD Related, ensure you are using an SD that is fully functional (and not too cheap), also try wiping your SD (or re-partitioning if you use Linux on your Switch) and starting again

fennectech, 2018-06-02:
Run a program called h2testw to see if you have gotten yourself a fake sdcard.     http://sushifury.com/2010/03/i-got-sold-a-fake-usb-flash-drive-on-ebay-heres-how-i-fixed-my-situation-and-how-you-can-avoid-this-situation-yourself/   If it reports your card is lying to you then that’s the cause of your problem.   Toss the card in the Trash because it WILL corrupt your data.

closed 2018-06-02 by SciresM

thomleg50, 2018-06-02:
yes i tried on +20 sd but i've got the same error

fennectech, 2018-06-02:
Did you try h2testw on the card   My 25$ card was fake.   Run h2testw and come back with the results

fennectech, 2018-06-02:
or buy THIS card https://smile.amazon.com/Sandisk-Ultra-Micro-UHS-I-Adapter/dp/B073JYVKNX/ref=sr_1_3?s=pc&ie=UTF8&qid=1527971222&sr=1-3&keywords=64+gb+micro+sd+card  It is from a trusted and known good seller