Atmosphere-NX/Atmosphere #116: Looking for an updated version of the hbmenu loader for 5.1.0

fennectech, 2018-06-02:
It was showin in this tweet https://twitter.com/sciresm/status/1002062352387260418?s=21. I’m willing to take the risk to get my save data restored. (I have it backuped up with checkpoint) but it got deleted in the game because of me reaching behind the tv to plug in my sound winding up pushing the right combination of buttons on my controller to erase my save.   Thanks for the help.

yellows8, 2018-06-02:
Build yourself -> https://github.com/switchbrew/nx-hbloader

fennectech, 2018-06-02:
THANK YOU FROM THE BOTTOM OF MY HEART!

fennectech, 2018-06-02:
Doesn’t load after compiling.  Same issue as the distributed build.

SciresM, 2018-06-02:
Not an issue in Atmosphere, and can't provide support at present.

closed 2018-06-02 by SciresM

fennectech, 2018-06-02:
Ok   if you get time id appreciate a copy of the one that works on 5.1.0