Atmosphere-NX/Atmosphere #124: Microsoft buying github.   Do we leave or stay?

fennectech, 2018-06-04:
I have been watching the news on github being bought and watched a few videos.  Gitlab would be a great alternative and has the ability to import complete projects including all the commit history issues notes projects and wiki. (It’s basically exactly like github but not M$)   The import is also verry straight forward and really doable with a single click.

theclashingfritz, 2018-06-04:
I suggest a leave. We have no clue what Microsoft will do and could fuck everything up before it could be saved.

SciresM, 2018-06-04:
I don't think we really have any problem with MS -- there are no indications at present that they're going to compromise GitHub's integrity as a platform. 

If GitHub starts degrading, I think migrating to an alternative service could make a lot of sense -- I don't think it makes sense to migrate just because of the announcement.

closed 2018-06-04 by SciresM