Atmosphere-NX/Atmosphere #128: Suggestion. Implement hook from error screen to reboot atmosphere on app crashes

fennectech, 2018-06-05:
I suggest adding a way to restart atmosphere after a homebrew crashed to keep users from having to reenter RCM mode every time an app crashes

jowadmax, 2018-06-05:
I believe that feature was implemented and then was stubbed out because of unresolved issues.

If you wait bit more until the first release, I bet it'll have automatic reboot to RCM.

fennectech, 2018-06-05:
That was the idea.  Just a suggestion for later development.    Automatic reboot into rcm isn’t super convent tho.  I don’t have an android device only a crappy windows PC that’s falling apart

SciresM, 2018-08-02:
On non-fatal error, Atmosphere is fine. On a fatal error, Atmosphere can't exactly do much anyway....the system reboots, so if the user has AutoRCM installed RCM will launch.

We may come back to this at a later time, as it's a good idea if reasonably implementable. It would require a custom fatal error module.

closed 2018-08-02 by SciresM