Atmosphere-NX/Atmosphere #135: Open source RCM dongle

fennectech, 2018-06-10:
I’m looking for a way to make an open source rcm dongle with OTS (off the shelf) parts that can also charge my switch.   I want to integrate it into this nyko powershell.   Ive got room for a teensy size board.   If i can get one with a right angle usbc cable soldered on it 
[image]

  The included cable has no data pins but the button can turn off the power  (which happens when the case is closed) and the other button turns it on.    This means i don’t need power management in the thing. But it would be nice if it only responded by sending payload (and using power) when an switch in RCM mode is detected.   Do you guys think an rcm dongle can also charge a switch that’s running horizon?

fennectech, 2018-06-10:
Do you think this would be possible?

Ninoh-FOX, 2018-06-10:
I maked this, but it doesn't charger the console.

https://fotos.subefotos.com/e14ee8d87d058949faddb9fde3aeeb81o.jpg

Kozlowlw, 2018-06-10:
What did you do to that poor trinket...

drofdoom, 2018-06-10:
I don't think so either forum or discords maybe and there's plenty of like minded projects .

hexkyz, 2018-06-10:
This is not an actual Atmosphere issue, so I'm closing it. However, feel free to drop by IRC or Discord for exchanging ideas with other like minded people who have been working on similar stuff.

closed 2018-06-10 by hexkyz

fennectech, 2018-06-10:
cna you link the discord?

fennectech, 2018-06-10:
Thanks    Sorry for clogging up atmosphere issues with it    I couldent find the discord for the group

jonirrings, 2018-06-14:
@fennectech https://discordapp.com/invite/C29hYvh

is homebrew discord, had seen a hardware channel, is that the right place?