Atmosphere-NX/Atmosphere #139: error when try to build

thomleg51, 2018-06-13:
i download newset source but unable to compile new sartophese because i got error .
can help me please and solution without closing ?

adamnock, 2018-06-13:
You need to stop. Your interference is not needed, wanted, or valuable in any way. Please leave and do not return

Comput3rUs3r, 2018-06-13:
I don't think this is the place to learn to program however I think adam is being a cock. :)

adamnock, 2018-06-13:
There’s a history with this guy making similar harassing posts and releasing ‘builds’ of the project publically

He was banned as thomleg50
And is thomhack on gbatemp, same problem

Cock level comment is warranted :)

Comput3rUs3r, 2018-06-13:
very well then, proceed. :+1:

simonmkwii, 2018-06-13:
You can't compile "sartophese" yet because two necessary ipc commands have not yet been implemented into libnx.

thomleg51, 2018-06-13:
@simonmkwii how can implement ? do you make tutorial for us ?
i see on discrod @SciresM do with path " rom sf " on oddyey . how do that ?

simonmkwii, 2018-06-13:
Are you that damn impatient?
I want romfs modding too, but you're acting as if the world depends on it.
**WAIT FOR THE OFFICIAL IMPLEMENTATION.**

P.S. I swear to god, don't you DARE @ me again thomleg51.
I will not have you clogging up my emails with your rectally dispensed "issue".
Someone close this shit.

drofdoom, 2018-06-13:
It's not ready to compile anyways .

foskon, 2018-06-13:
Stop being annoying and wait.