Atmosphere-NX/Atmosphere #142: error compile mitm disord

thomas-50, 2018-06-14:
there are syntax error in discord mitm by @SciresM please listen do not close .

log paste :https://pastebin.com/uVMXxKDy

SapphireExile, 2018-06-14:
Why do you insist on opening issues continuously? If Scires had a syntax error, he would fix it. Not tell you to open your thousandth issue.

borntohonk, 2018-06-14:
you again? lol

foskon, 2018-06-14:
Just ignore him. And, if you can, close.

simonmkwii, 2018-06-14:
At this point I'm starting to think there's a syntax error in your brain.
I'm assuming you're referring to this, this was reposted in my server:
[image]
What part of "needs a lot of testing" do you not understand?

closed 2018-06-14 by TuxSH