Atmosphere-NX/Atmosphere #148: LayeredFS function on System titles?

torunado, 2018-06-20:
I tested layeredFS function on few games, but it didn't work on system titles.
Is it just becuase Atmosphere doesn't support that?

simonmkwii, 2018-06-20:
Yeah, it would be nice!
I want to replace the album icon with a homebrew one 😄

darvell, 2018-06-27:
Pure theory but I think the MitM is registered too late to handle any of the system titles. Also UI elements aren't part of the system titles per se and more part of the OS itself.

thomasnet-mc, 2018-07-02:
Try with modifying the function should_mitm (it's an inline so in a header file, don't remember which one).

parrotgeek1, 2018-07-03:
it's stratosphere/fs_mitm/source/fsmitm_service.hpp Line 23 
that's definitely why it's not working.

simonmkwii-dev, 2018-07-11:
@darvell Considering it's run as a KIP (kernel initial process), it gets registered during kernel initialisation, therefore any sysmodule or sysapplet with a RomFS section (with TID >= 0x0100000000000006) should be patchable.

SciresM, 2018-08-02:
Support was added for this in  https://github.com/Atmosphere-NX/Atmosphere/commit/cd42f6dc18347e5e5a967c298c6caf4efd1e440c

closed 2018-08-02 by SciresM