Atmosphere-NX/Atmosphere #154: Stratosphere fails to build

dzervas, 2018-06-23:
I'm on latest devkitpro and when I try to compile everything, I get:

```
aarch64-none-elf-g++ -MMD -MP -MF /source/stratosphere/loader/build/ldr_shell.d -g -Wall -O2 -ffunction-sections -march=armv8-a -mtune=cortex-a57 -mtp=soft -fPIE -DDISABLE_IPC -I/source/stratosphere/loader/include -I/opt/devkitpro/portlibs/switch/include -I/opt/devkitpro/libnx/include -I/source/stratosphere/loader/../libstratosphere/include -I/source/stratosphere/loader/build -D__SWITCH__ -fno-rtti -fno-exceptions -std=gnu++17 -c /source/stratosphere/loader/source/ldr_shell.cpp -o ldr_shell.o 
In file included from /source/stratosphere/loader/../libstratosphere/include/stratosphere.hpp:3,
                 from /source/stratosphere/loader/source/ldr_shell.cpp:2:
/source/stratosphere/loader/../libstratosphere/include/stratosphere/ipc_templating.hpp: In function 'Result WrapIpcCommandImpl(Class*, IpcParsedCommand&, IpcCommand&, u8*, size_t)':
/source/stratosphere/loader/../libstratosphere/include/stratosphere/ipc_templating.hpp:512:11: error: 'struct IpcParsedCommand' has no member named 'IsDomainMessage'
     if (r.IsDomainMessage) {
           ^~~~~~~~~~~~~~~
In file included from /source/stratosphere/loader/../libstratosphere/include/stratosphere/ipc_templating.hpp:544,
                 from /source/stratosphere/loader/../libstratosphere/include/stratosphere.hpp:3,
                 from /source/stratosphere/loader/source/ldr_shell.cpp:2:
/source/stratosphere/loader/../libstratosphere/include/stratosphere/isession.hpp: In member function 'virtual Result ISession<T>::handle_message(IpcParsedCommand&)':
/source/stratosphere/loader/../libstratosphere/include/stratosphere/isession.hpp:105:19: error: 'struct IpcParsedCommand' has no member named 'IsDomainMessage'
             if (r.IsDomainMessage && this->active_object == NULL) {
                   ^~~~~~~~~~~~~~~
/source/stratosphere/loader/../libstratosphere/include/stratosphere/isession.hpp:110:19: error: 'struct IpcParsedCommand' has no member named 'IsDomainMessage'
             if (r.IsDomainMessage && r.MessageType == DomainMessageType_Close) {
                   ^~~~~~~~~~~~~~~
/source/stratosphere/loader/../libstratosphere/include/stratosphere/isession.hpp:110:40: error: 'struct IpcParsedCommand' has no member named 'MessageType'
             if (r.IsDomainMessage && r.MessageType == DomainMessageType_Close) {
                                        ^~~~~~~~~~~
/source/stratosphere/loader/../libstratosphere/include/stratosphere/isession.hpp:110:55: error: 'DomainMessageType_Close' was not declared in this scope
             if (r.IsDomainMessage && r.MessageType == DomainMessageType_Close) {
                                                       ^~~~~~~~~~~~~~~~~~~~~~~
/source/stratosphere/loader/../libstratosphere/include/stratosphere/isession.hpp:110:55: note: suggested alternative: 'DomainMessageHeader'
             if (r.IsDomainMessage && r.MessageType == DomainMessageType_Close) {
                                                       ^~~~~~~~~~~~~~~~~~~~~~~
                                                       DomainMessageHeader
/source/stratosphere/loader/../libstratosphere/include/stratosphere/isession.hpp: In member function 'Result ISession<T>::handle_signaled(u64)':
/source/stratosphere/loader/../libstratosphere/include/stratosphere/isession.hpp:186:32: error: 'struct IpcParsedCommand' has no member named 'IsDomainMessage'
                         if (!r.IsDomainMessage || r.ThisObjectId >= DOMAIN_ID_MAX) {
                                ^~~~~~~~~~~~~~~
```

fincs, 2018-06-23:
devkitA64. You're on latest devkitA64. devkitPro is not software, it's the organization that makes devkitA64 and many other things.

It looks like right now Stratosphère requires switchbrew/libnx#112 to be merged.

Benoit7413, 2018-06-26:
You just have to clone libnx, build it and replace it

hexkyz, 2018-06-26:
Stratosphère is dependant on libnx. New changes will often cause building issues until the necessary changes are reviewed and merged in libnx.

closed 2018-06-26 by hexkyz