Atmosphere-NX/Atmosphere #155: aarch64-none-elf-gcc: error: unrecognized command line option '-mthumb'

intraum, 2018-06-24:
trying to build on macOS, coming across this error; someone else had this problem which apparently was fixed by pointing DEVKITARM to devkitARM, but as you can see, that's what i've done, to no avail.
```
$ make DEVKITPRO=/opt/devkitpro/ DEVKITARM=/opt/devkitpro/devkitARM

/Library/Developer/CommandLineTools/usr/bin/make -C fusee all
/Library/Developer/CommandLineTools/usr/bin/make -C fusee-primary all
chainloader.c
aarch64-none-elf-gcc -MMD -MP -MF /Users/kennedy/Downloads/Atmosphere-master/fusee/fusee-primary/build/chainloader.d -g -O2 -fomit-frame-pointer -ffunction-sections -fdata-sections -std=gnu11 -Werror -Wall -fstrict-volatile-bitfields -march=armv4t -mtune=arm7tdmi -mthumb -mthumb-interwork -D__BPMP__ -DFUSEE_STAGE1_SRC -I/Users/kennedy/Downloads/Atmosphere-master/fusee/fusee-primary/include  -I/Users/kennedy/Downloads/Atmosphere-master/fusee/fusee-primary/build -c /Users/kennedy/Downloads/Atmosphere-master/fusee/fusee-primary/src/chainloader.c -o chainloader.o 
aarch64-none-elf-gcc: error: unrecognized command line option '-mthumb'
aarch64-none-elf-gcc: error: unrecognized command line option '-mthumb-interwor'
make[3]: *** [chainloader.o] Error 1
make[2]: *** [build] Error 2
make[1]: *** [fusee-primary] Error 2
make: *** [fusee] Error 2
```

fincs, 2018-06-24:
You seem to have somehow installed devkitA64 to $DEVKITPRO/devkitARM. The recommended course of action is to completely delete your current installation, then start over by following the official [Getting Started](https://devkitpro.org/wiki/Getting_Started) guide. You need to install the `switch-dev` group as well as the `devkitARM` package.

intraum, 2018-06-24:
fixed! thanks, dunno how i managed to do that

closed 2018-06-24 by intraum

jme2712, 2019-03-21:
After having the above issue I'm now struggling with something similar. Here is what I get. 

$ make DEVKITPRO=/opt/devkitpro/ DEVKITARM=/opt/devkitpro/devkitARM
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
/Library/Developer/CommandLineTools/usr/bin/make -C thermosphere all
Makefile:10: /opt/devkitpro//devkitA64/base_rules: No such file or directory
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
make[1]: *** No rule to make target `/opt/devkitpro//devkitA64/base_rules'.  Stop.
make: *** [thermosphere] Error 2

shchmue, 2019-03-21:
install devkitarm-rules via dkp pacman