Atmosphere-NX/Atmosphere #159: Idea to prevent horizon from logging to disk

fennectech, 2018-06-27:
I know that on many platforms creating a directory where the os wants to put its file or a file where the os wants to put a directory will prevent the file from being created.   So if we use that technique it may prevent the switch OS from logging. But i dont have any way to test it myself.       But say the switch puts a file at /user/log/file.log you create a directory with the name file.log in its place the switch may not delete it on its own.     The file may just stay there and then no logging could occur.      Sorry that this isn’t appropriates here but i dunno where to put it.

drofdoom, 2018-06-27:
Don't work .

fennectech, 2018-06-27:
Damn.   Was a good idea

closed 2018-06-27 by fennectech