Atmosphere-NX/Atmosphere #160: Question/Suggestion: Add Bluetooth support for third Party devices

GitHubStefan13, 2018-06-29:
The Switch itself does not allow us to add Bluetooth Headsets or the Steam Controller for example.

My Question is if libnx or Atmosphere already allow this?

And my Suggestion is to include a Bluetooth Driver in Atmosphere if not already planned.

TuxSH, 2018-06-29:
> And my Suggestion is to include a Bluetooth Driver in Atmosphere if not already planned.

Probably a baremetal driver at a later time.

> The Switch itself does not allow us to add Bluetooth Headsets or the Steam Controller for example.

Not yet, and supporting these devices after the OS has booted may need even more work.

victorheld, 2018-06-29:
I also would like this eventually, but I understand that it is not really a priority right now

GitHubStefan13, 2018-06-29:
I totally get that its not a priority right now.

The main reason i was curious about it was because i am interested in "trying" to do it myself (which will take a lot longer).

I would appreciate it if you could lead me in the right direction though, since im not really experienced with the whole libnx library and atmosphere.

evega2, 2018-06-30:
What third-party bluetooth accessories are you trying to connect to the switch anyway?

GitHubStefan13, 2018-06-30:
@evega2 Read my Question again, i stated two devices already.

But to make it clear again, the Switch only allows Devices from Nintendo(as far as im concerned) to connect to the Switch via BL.

In case you want to connect BL-Headsets or Third Party Controllers that work over BL for examle the Steam Controller or Xbox One S Controller.

Furthermore you could connect also connect a Keyboard and Mouse and pretty much all devices that work over BL.

Well thats my plan anyway, how well that would work needs to be tested.

tiliarou, 2018-06-30:
It's a nice idea but of course not priority.

fennectech, 2018-06-30:
I would LOVE native support for my xbox one controller!       I currently use a mayflash adapter but that only works in the dock.  PLUS headset support would be the holy grail!

evega2, 2018-06-30:
@GitHubStefan13 Yeah sorry, my bad. A wireless keyboard would be great and also controllers without the dongle.

fennectech, 2018-06-30:
So is this something well get “eventually”.   Or. Most likely not

Zero-machine, 2018-07-10:
You can use a Magic NS Adapter in the meantime, it supports all the main bluetooth controllers and can be used both in the docked and tablet mode, although this last one needs also a cable connected to the dongle. https://www.amazon.com/Mayflash-MAGIC-NS-Wireless-Controller-Fighting-PC/dp/B074JZPS39#customerReviews

SciresM, 2018-08-02:
This issue is out of scope, for now. In a year or two, perhaps come back -- you're also welcome to implement, and I would accept a pull request ;)

closed 2018-08-02 by SciresM

galli-leo, 2019-01-07:
@SciresM Hi there, I would like to implement this (or at least try). Do you have any pointers to begin doing this? From my short research it seems like I would have two options:

1. MITM the Bluetooth Driver Service and return other devices / pair other devices / report data for other devices
2. MITM the HID Service, but I am not sure how exactly that relates to the controllers via bluetooth.

Am I correct in the assumption, that the controllers are managed over the bluetooth driver service? Do you know where there are some resources of the inner workings of the switch system / where I could get a full dump to analyze for code? (Or an easy way to dump them myself). Do you know if there has been an work done on reverse engineering how the controllers are handled by the firmware?

linxucc, 2020-10-18:
Found this old talk when I'm about to create a new issue, I'm also interested in the bluetooth support on the switch, mostly to the bluetooth headset support part. Curious anyone still into this? 

Is this thing possible but needs a lot of hard work, or it's just proved to be impossible or near impossible?

For now the Atmosphere is already in a near perfect form (at least in my common folk's eye), maybe time for the bluetooth thing? 

The bluetooth earbuds and headset has boomed in the past few years, everyone use them now, and the incompatibility to use bluetooth earphone has become a main complain through users, and it's my most pain on the switch... (and PS4, it's silly)

Could someone shed some light on me, like, if it's possible, what's the general path to do this? Or there're already some people in the community looking into it?

It's hard to find any useful information through the google search, so I'm asking here...

Thanks!

masagrator, 2020-10-18:
You need to implement bluetooth audio codes because on HOS it doesn't exist. And still this feature is pointless for most of games Because it's not possible to implement low input codec.

MarioMasta64, 2020-10-19:
@GitHubStefan13 in the case of controllers [Mission Control](https://github.com/ndeadly/MissionControl/) works nicely and is slowly adding more support

linxucc, 2020-10-19:
Wow Thanks! It’s great information... 
“Mission control” looks pretty cool, definitely will try it on my other controllers, and the project’s document also well explains how it’s working and why audio is very hard to do... 
Maybe better off for a dongle to save the time and hassle... Thanks for all you guys, my quest is over now.😂