Atmosphere-NX/Atmosphere #162: NTFS support pheasable?

fennectech, 2018-07-02:
How difficult would it be to add NTFS support to atmosphere  I know it would be difficult but would it be a level of difficult to make it not worth it?     Exfat is just sooo fragile.   we need a journaling filesystem.

mariogamer2, 2018-07-02:
Honestly better implement ext* filesystem at this point. Ntfs is only useful for a few things, that ext* do better (in general). Only an opinion tho.

fincs, 2018-07-02:
ext\* is almost *impossible* to use on Windows. NTFS can be used on all 3 major OSes.

adamnock, 2018-07-02:
Why not go with something really obscure, like BeFS, so that everyone has difficulty with it and the project can help build global cooperation and lead to a better world?

mariogamer2, 2018-07-02:
Well I remember the wii days where you could have a disk/sd formatted in ext*/ntfs/*fat. But I guess that's not the same at all.

(It'd be cool if a switch could read all these tho...)

@adamnock :p

dmlemos, 2018-07-02:
Honestly `ext4` is the more compatible of the two. Let's see, `ext4` works on linux, android and bsd. `NTFS` is pretty much windows only. 

Besides `ext4` performance is better. To retain windows compatibility there is always `exFAT` already supported.

fincs, 2018-07-02:
NTFS is not "Windows only". NTFS is supported right out of the box in most popular Linux distros (read/write, ntfs-3g & fuse), as well as macOS (read-only, unless macfuse is used). ext4 simply does not work properly outside Linux. There do not exist stable free ext4 fs drivers for Windows **at all**.

fennectech, 2018-07-03:
Well NTFS-3g on linux provides adequate support (Read Write Repair and Create.).  While my linux box cannot repair an exfat drive and fat32 is just shite.  I say NTFS is the best option.   And I’m a *nix user.

fennectech, 2018-07-03:
Ext4 doesnt work on windows (personally not a problem for me  but compatibility is important for a community release).  Ive found exfat to be really fragile.   Without any journaling its really not tolerant to power failures or bad shutdowns (wich happen a lot on homebrew)    Plus it would be far more tollerant of app crashes that normally botch our exfat filesystems

fennectech, 2018-07-03:
How about fat12?  Or we can blindly write bytes to the disk so NOBODY can make sense of them. Not even us!

parrotgeek1, 2018-07-03:
@fennectech ^ I think that's called WBFS :-p

fennectech, 2018-07-04:
That was a huge mess.

fennectech, 2018-07-04:
Supposedly Apple improved NTFS support on macOS     I say NTFS Is the best.  It has wid reaching support.

SapphireExile, 2018-07-04:
ExFAT would be the best in terms of total compatibility. NTFS has issues on some linux distro (mounted read only you say?), while ExFAT is wholly compatible.

TuxSH, 2018-07-04:
ExFAT is supported.

Implementing support for anything that FAT* and ExFat is a waste of time since HOS doesn't support it anyway.

Also please use a spell checker -- the horrible spelling mistake in the title of your 'issue' already gives a bad impression itself.

closed 2018-07-04 by TuxSH

jonirrings, 2018-07-04:
fs support is a time wasting hardwork, 

I don't think there is any necessity to implement any other fs support other than fat32 and exfat.

fennectech, 2018-07-04:
THe reason is the fragility of bolth exfat and fat32.  Nether of them support journaling

jonirrings, 2018-07-05:
@fennectech just forget the fragility, even ninty does not care about it.