Atmosphere-NX/Atmosphere #166: SDCard issues with new SDCard Drivers(SDXC Related?)

me2151, 2018-07-21:
I get SDCard errors using the new SDCard Drivers with fusee-primary. The error I get is Failed to send OP Cond. I Believe this is due to my card being SDXC but I cannot confirm that as I do not have any SDHC Sdcards. I have tried 2 different SDXC SDCards, one is 128gb and one is 256gb. Both cards were tested as Fat32 and exFat.

Here is a photo of the error.

https://cdn.discordapp.com/attachments/461992992232964097/465315873264238592/20180707_203839.jpg

Let me know if you need any additional information.

hexkyz, 2018-07-23:
There was a bug in the SDMMC clock speed setup. Could you please try again with the latest commit?

me2151, 2018-07-23:
Yes I can try again In 6 hours(at work right now so no switch).

me2151, 2018-07-24:
Still Does not work. Same error

me2151, 2018-07-25:
@hexkyz Fixed with latest commit https://github.com/Atmosphere-NX/Atmosphere/commit/a2104b85ceb114aa104bddbad1871121305717fc

closed 2018-07-25 by me2151