Atmosphere-NX/Atmosphere #167: When will sig patches get implemented?

lunalik2, 2018-07-25:
Myself, and many others are currently using sig patches ripped from an older build of ReiNX, and putting them into atmosphere manually. I pretty much did this as a temporary solution to the lack of sig patches on the cfw, and shared it to others online. Any idea on when the sig patches will get implemented themselves? Thanks.

SapphireExile, 2018-07-25:
I may be wrong, but I'm pretty sure they wouldn't be implemnting this as it's piracy related. I'd personally like it as homebrew could be installed as a title.

mariogamer2, 2018-07-25:
I don't think it'll be implemented, and not even for homebrew, since SciresM want a framework for launching hb menu. (like luma3ds)

fincs, 2018-07-25:
My 2 cents: As it stands right now, NCA installation = Piracy. "Homebrew" NCAs can technically happen but I see zero benefit in them vs homebrew NROs loaded through HBL, and I only see a plethora of problems regarding custom titles (especially with Nintendo banning users who have installed custom titles with or without invalid title IDs). I think we would be better off if people would stick to NRO. There's really nothing we can't change in HBL, and FS should already have full permissions, as well as full service permissions (without even needing smhax, which was reimplemented anyway). Homebrew NCA per se aren't evil, but they involve too many potential issues and inconveniences to be a good way to run support homebrew.

Comput3rUs3r, 2018-07-25:
I think you should make your own fork of atmosphere with the things that 90 percent of the people want.
thanks Lunalik.

lunalik2, 2018-07-25:
@Comput3rUs3r np. I don't know if I'll be able to release this on github though. I has parts of the ES, so it is kinda in the grey area. It even got removed from ReiNX because of this. @SapphireExile well if you are interested in using them for development, or personal use, search for Lunalik Person on Facebook, and check my first and only post.

drofdoom, 2018-07-25:
I think without sigs being patched its not a cfw just a mfw but im old school i guess .

mariogamer2, 2018-07-25:
@drofdoom  Other consoles had it in general because of it being bether (PlayStation particularly). The 3ds mostly had it for homebrew, cause a good framework was implemented very late. The wiiu: well idk. No one cares apparently :/.
As fincs explained, too many drawbacks, and mostly for piracy, so for now, 0 benefits.

lunalik2, 2018-07-25:
@mariogamer2 I used it to install Hulu for devMenu. I also used it to install backups of games I OWN (so I don't have to bring the carts everywhere). There are uses other than piracy, and there will be more once hb apps are developed.

mariogamer2, 2018-07-25:
Yeah I know that installing backup of games is legit. But still.

(off topic: Hulu is on the switch?)

rajkosto, 2018-07-25:
You can install backups of games and updates you dumped from that console without any "sigpatches".
Anyone asking for these is strictly for piracy (as they arent asking for unsigned NCA loading either, they are asking for patched ES to install fake tickets.)

mariogamer2, 2018-07-25:
@rajkosto But I don't think this is cartdridge related, right?

rajkosto, 2018-07-25:
You can install anything you own without forged tickets.

mariogamer2, 2018-07-25:
Ok.
I thought it wasn't related cause of the cartridge specific stuff.

AnalogMan151, 2018-07-25:
@fincs Will we ever be able to suspend homebrew? Since we load it through an applet this isn't currently possible.

Reisyukaku, 2018-07-25:
Some people just want to play their title key encrypted homebrew games. ( ͡° ͜ʖ ͡°)

fincs, 2018-07-25:
@AnalogMan151 Keep in mind that supplanting the album applet isn't the only option. Any other legitimate title (i.e. non-applet too) should be possible as the HBL host title.

@Reisyukaku None exist atm and as I explained previously, there are no plans to support homebrew NCAs at the moment, due to lack of benefit. As it stands right now, titles with forged keys equate to piracy.

Reisyukaku, 2018-07-25:
I was being facetious @fincs :p 
title key crypto is gross anyways.

drofdoom, 2018-07-25:
@Rei you get it ,its not a true cfw without it I dont care what anyone says its just moded sys modules .

fennectech, 2018-07-25:
i could see it being legit as consoles that have been banned from the CDN cannot redownload their own content leaving downlaoding it being the only option for playing legitimate content

mariogamer2, 2018-07-26:
@fennectech they used cdn first -> let's not be blind, most people exlcluding devs use it for piracy.

(off topic: I do see that case as legitimate. It hapenned to me with mario kart, the game was for no reasons unlayable. I was stupid enough to buy it again.)

LittleWhite-tb, 2018-07-26:
My thoughts :
- Atmosphere is expected (and was even more expected before ReyNX). People wants piracy even if they have to buy the Xecuter stuff. But, first, I don't like what they have done (it's an opinion like others, and it's mine). I don't like how it's closed source (and this is even worse) and I don't like how they can brick console if they detect this or that. For that, I would prefer to choose an open source solution and as a developer, I would prefer to offer the opportunity to users to not go to such product like SX.
- (unsure about this one) I would prefer to have HBL directly in the home menu like any other application (same for others homebrews). Going to the album looks hacky. I mean, when I have take the handle on my console, I prefer to have something neat. 3DS homebrew integration is cool for that. Once you have the CIA, it's in the home menu and looks like any other application.

Of course, I can also understand your pros and cons, and I have followed this discussion with interest. Here was just my personal thoughts.

rajkosto, 2018-07-26:
You can have hblauncher as menu icon without any patches, just exefs redirect of flog or similar system application.

drofdoom, 2018-07-26:
Not neat enough thats just a work around plus wouldn't it have to be a pure exefs title ?

drofdoom, 2018-07-26:
I meant the donor

rajkosto, 2018-07-26:
what romfs the donor has is irrelevant when the redirected exefs does not mount it.

drofdoom, 2018-07-26:
Fair enough then why suggest flog and system titles with pure exefs ?

SciresM, 2018-08-02:
Closing this for now.

closed 2018-08-02 by SciresM