Atmosphere-NX/Atmosphere #168: Suggestion: APK Wrapper

HyrulianTerror, 2018-07-27:
Eventually I would like to see support for android applications (not Bluestacks, but compatibility layer), like WINE for Linux, but for android. This would be excellent for voice chat through Discord or Nintendo switch online.

aveao, 2018-07-27:
Not a reswitched/atmosphere dev, but I feel like this'd be tricky to implement.

Most such emulators for android are developed in a proprietary fashion by companies (Google, Microsoft, Genymotion, Bluestacks etc), and even if you somehow found a code for an emulator that works well enough, getting a compatibility layer would still take a lot of effort due to the differences on the OS and the kernel. 

Wine project is one that's been in development for 25 years now, and it's still not perfect.

I feel like this is a case of the [XY problem](https://meta.stackexchange.com/questions/66377/what-is-the-xy-problem). Just writing a homebrew discord application or porting nintendo switch online as a homebrew app would be simpler in this case (though good luck getting discord to run in background).

fennectech, 2018-07-27:
This would be INCREDIBLY dificult to implement.  It’d be way easier if we were on a ‘*nix system becuas we could just chroot.  But on horizon this would be nearly impossable.  Its more likely that we will get a port of android for the switch that runs on bare hardware instaid of in horizon.

uyjulian, 2018-07-27:
Running Yuzu in Linux would be much easier than this suggestion.

TuxSH, 2018-07-27:
Not feasible at all.

closed 2018-07-27 by TuxSH