Atmosphere-NX/Atmosphere #173: Fix remaining bugs preventing Exosphere boot.

SciresM, 2018-08-02:
Exosphere still does not boot some firmwares. Determine why, and fix.

encounter, 2018-08-10:
For 4.0.1, #169 explains my experience with it. As far as I can tell, Exosphere boots but freezes the BPMP in the process. Haven't debugged further.

encounter, 2018-08-19:
Exosphere appears to launch, but afterwards I get a black screen on 4.0.1 and 5.1.0.

tumGER, 2018-08-19:
Pretty sure that has nothing to do with exosphere - Try launching exosphere with hekate (secmon=exosphere.bin) @D3fau4

pplatoon, 2018-08-19:
after loading everything well, it stays black,

i m on 3.0.0

aveao, 2018-08-19:
I'm having a very bizarre issue.

When I close Skyrim (latest version, from eshop), my switch shuts down completely right after it successfully closes (game close animation plays, returns to main menu indicating success, then shuts down).

The horizon shut down animation doesn't play, a regular switch error does not happen, it just goes black. When I press power again it gets me into RCM as I have AutoRCM (note: it sometimes gets me into autorcm, it sometimes does a weird thing where it does neither). I tested this with creport and without sys-ftpd. creport doesn't create any files on sdcard after this.

When I try doing the exact same thing with [tomGER's Atmosphere patches (which disable exosphere)](https://github.com/tumGER/SDFilesSwitch/blob/master/Resources/Patches/atmosphere.patch), it works perfectly fine. All other games I own are fine with exo btw.

On 5.1.0+exfat with builds of atmosphere on latest stable libnx (1.3.2). I use latest dev hekate source code as a bootloader, and I use all kips except boot and fs_mitm.

---

Boring, socially anxious, shaky, out-of-focus video: https://youtu.be/ADz3oQnjSjI

Exact ATM builds I used: [Patched](https://bsnx.lavatech.top/atmosphere/atmosphere-patched-e5bfb95.zip), [Unpatched](https://bsnx.lavatech.top/atmosphere/atmosphere-unpatched-e5bfb95.zip)

Exact hekate build I used: [Dev branch at 5fd9daa](https://bsnx.lavatech.top/hekate-ctcaer/hekate-ctcaer-dev-5fd9daa.zip)

Relevant parts from hekate-ipl.ini:

```ini
[Atmosphere]
secmon=modules/atmosphere/exosphere.bin
kip1=modules/atmosphere/loader.kip
kip1=modules/atmosphere/sm.kip
kip1=modules/atmosphere/pm.kip
atmosphere=1
[Atmosphere - Exo]
kip1=modules/atmonoexo/loader.kip
kip1=modules/atmonoexo/sm.kip
kip1=modules/atmonoexo/pm.kip
atmosphere=1
```