Atmosphere-NX/Atmosphere #180: Create and fill out a GitHub Wiki for Atmosphere.

SciresM, 2018-08-02:


TheFirstAv3nger, 2018-08-06:
I'd be glad to write that up for you guys.

Resaec, 2018-09-18:
I'm not all too good at this, but [here is a basic one](https://github.com/Resaec/Atmosphere/wiki/).
I will extend it so there are as few red pages in the sidebar as possible.

It does not meet any expectations but it could give someone an idea how to start it off.

iirc there is no PR mechanic for wikis but there is ... .wiki.git

Resaec, 2018-09-22:
Nice! I didn't know wiki is free to edit to everyone when not disabled manualy.
Still many things left to add.

roblabla, 2018-09-23:
I really dislike the use of github wiki for this.

- Github wiki has no PR system.
- The contribution system is either "anyone can edit" or only contributors can, making collaboration with newcomers painful.
- It has no visible history accessible from github

I believe having the docs in a dedicated docs/ folder, with files in Markdown, to be a superior option.

Resaec, 2018-09-23:
It has been a wish by the community since ages...

noirscape, 2018-09-23:
@roblabla While your other points remain valid, the history of a wiki page _is_ accessible through GitHub.

https://github.com/Atmosphere-NX/Atmosphere/wiki/Home/_history

HylianMedia, 2018-09-24:
@joelc608 and I created a first draft of the Wiki [here](https://github.com/joelc608/Atmosphere/tree/master/docs). We started with the basic wiki from @Resaec and built upon it, so huge thanks to him and everyone that helped us on the Discord!

joelc608, 2018-09-24:
> 
> 
> @joelc608 and I created a first draft of the Wiki [here](https://github.com/joelc608/Atmosphere/tree/master/docs). We started with the basic wiki from @Resaec and built upon it, so huge thanks to him and everyone that helped us on the Discord!

I put in a pull request for this.

Resaec, 2018-09-24:
Linking issue and PR #210

hexkyz, 2018-10-05:
PR #210 has been merged and basic documentation is now available under the docs/ folder using the Markdown format.

closed 2018-10-05 by hexkyz