Atmosphere-NX/Atmosphere #184: Feature: Multi-LayeredFS

AnalogMan151, 2018-08-06:
TL;DR: Allow for multiple of the same romfs folder containing ROMFS files to be loaded in descending order using a delimiter naming system.

This request would be best explained with an example. Take using LayeredFS to mod the Legend of Zelda: Breath of the Wild. 

In this example we will install multiple mods, then go over the options for removing just one. 

Mods included: Linkle Mod, Normal Paraglider for Linkle Mod, Unbreakable Weapons and Items mod, Infinite MasterCycle Fuel mod, Enhanced Wolf Link mod.

Right now the current system would be to copy all these mods into the `01007ef00011e001\romfs\` folder, overwriting any files when prompted (making the order you add them in important) so they are all mashed together.

Now, let's try and remove just the Unbreakable Weapons mod. This mod has 42 directories and 1128 files. The first option would be to go through each folder and remove the files related to this mod (YUCK!). The other option is to simply delete the entire `romfs` folder and start all over with adding in the Linkle mod, Normal Paraglider for Linkle mod, Infinite MasterCycle Fuel mod, and Enhanced Wolf Link mod. Neither option is entirely ideal.

My proposed feature request would allow for something like this:
```
titles
└── 01007ef00011e000
    ├── romfs_01LinkleMod
    ├── romfs_02NormalParaglider
    ├── romfs_03UnbreakableWeapons
    ├── romfs_04InfiniteFuel
    └── romfs_05EnhancedWolfLink
```
When LayeredFS reads the folders in the title ID, it will split the folder names by the `_` delimiter and use the second part to look for files in descending order. Essentially it starts loading files from the last alphabetical folder and works it way up through each folder until the file requested is found than stops.  Ideally it only parses one delimiter, so that you can also have naming schemes like: `romfs_01_Linkle_mod` for easier readability. 

This would allow for a multi-layered redirect system making it easier on the end user to stack and remove mods as well as change their loading order. 

I do not know how much this would impact loading times though, so if this is technically infeasible, perhaps someone can propose a similar working idea that accomplishes the same goal but has less overhead.

takashi1kun, 2018-08-16:
This can be solved whit a homebrew mod manager, have all the mods uncompresed on individual folders, then you set a load order in the mod manager and copy the files to the romfs folder in that order, when you want to change it, it remove everything on that romfs folder and copy the mods in the new order, is shitty but it works, another option would be virtual link-ish soultion like some skyrim mod managers for pc do.

AnalogMan151, 2018-08-16:
That is certainly one way, but using a homebrew is outside the scope of this request, which is to have the multi-layer feature built into the CFW. Using a homebrew would require keeping copies of the mods in their separated form while they get merged together in the atmosphere folder. 

Having them handled natively would actually be even easier for a homebrew to manage as it only needs to copy the new mod into the folder and name it so that it enters in the order you want. You can even move mods up or down the loading order my just renaming the folders.  If anything, it makes it easier for the end user as well as homebrew makers.

Resaec, 2018-09-17:
I think this is a good idea. It enables multiple mods to be loaded and reduces conflict potential, even tho the mods need to be "multimod able"
But I would sort them differently to make it easier to remove single mods and keep them contained in their own directories.
```
atmosphere/
└── titles
    └── 0123456789
        ├── 01_patch1
        │   ├── exefs
        │   └── romfs
        ├── 02_patch2
        │   └── exefs
        └── 10_UnbreakableWeapons
            └── romfs
```

SciresM, 2019-05-23:
Wasn't crazy about this suggestion a year ago, still not crazy about it.

We're theoretically open to something along these lines, but I don't think the system described is the one we'd want to use.

closed 2019-05-23 by SciresM

DiamantTh, 2022-09-22:
Hello,
Is there any news on how this will be implemented?
What approaches are there actually for this scenario?
Because especially if there are several mods from different creators and these also create for the simple user.

Yuzu supports this kind of mods yes, no idea if there were big problems of some kind in the development.

Should actually since this is closed a new issue be opened in which we can discuss?
Greetings

SciresM, 2022-09-22:
Discussion is best held on the ReSwitched discord, not on GitHub.