Atmosphere-NX/Atmosphere #185: Feature: Per-user LayeredFS

AnalogMan151, 2018-08-06:
Since the Switch is much more family oriented than individual oriented like the 3DS was, having mods be loaded based on profile played would be an appreciated feature. Something as simple as adding a username to the end of a title folder like so:

```
titles
├── 01007ef00011e000_User01
│   └── romfs
└── 01007ef00011e000_User02
    └── romfs
```

If a folder has no user name after it then the mod is loaded for all user accounts. This would allow one to add mods only for their own profile or add different mods based on who loads the title.

Luro02, 2018-08-08:
this would cause problems, if someone has their name in for example kanji....so using (alternatively) the account ID should be an option too ?!

griest024, 2018-08-08:
This would certainly be helpful but it wouldn't have to be implemented in CFW. You could accomplish essentially the same behavior with some good homebrew.

CobaltTiger, 2018-08-08:
As I recall, there was a homebrew app on the 3DS that did exactly that, minus the multi-user thing. Conceptually, it's really similar, so I wonder if it couldn't just be ported?

griest024, 2018-08-09:
It might work out fine for some minor content but for LFS files that change major things (i.e. the executable), I can't see how that could be dynamically injected and actually work as intended.

AnalogMan151, 2018-08-16:
While a homebrew can certainly do this, it's outside the scope of this request. This is for benefitting the end users, especially those who are not familiar with homebrew. With the Switch being a household unit, there will be instances were the people playing it are not the person who hacked it and thus will not know to to first open a homebrew and adjust the installed mods. Having it built right into the CFW would be ideal for a non-technical person's standpoint. 

Plus, if this is implemented it would make creating a homebrew to manage per-user mods much easier.

ThatNerdyPikachu, 2018-08-20:
I'll implement this, look out for a PR soon...

ThatNerdyPikachu, 2018-08-20:
(only thing i may not be able to do is the part where one folder gets loaded always..)

AnalogMan151, 2018-08-20:
For clarification, I didn't mean both a universal folder and also a user folder getting loaded, I mean, check to see if a user specific folder exists and load the romFS from that, if it doesn't, then check to see if a non-user specific folder exists, then load that. Kind of like a fall back.

ThatNerdyPikachu, 2018-08-20:
@AnalogMan151 Oh, that's what I thought you meant initially. Thanks for the clarification!

I'll PR later today.

SciresM, 2019-01-26:
This turns out to be basically not feasible due to a lack of good way to represent users inside filepaths (user names not unique, user ids inconvenient and unintuitive, special characters are a problem, etc)

closed 2019-01-26 by SciresM

bakugo, 2019-01-26:
> user ids inconvenient and unintuitive  

How are user ids more inconvenient or unintuitive than title ids?

SciresM, 2019-01-26:
@shchmue user IDs are actually 128 bits.

AnalogMan151, 2019-01-26:
True. Unless we did something that like matching only the first 32 or 64 bits. It’s unlikely two accounts on one Switch would have similar enough IDs to have multiple matches.

SciresM, 2019-01-26:
We're not using user-ids.

SciresM, 2019-01-26:
@roblabla as mentioned, usernames are not unique

roblabla, 2019-01-26:
Is that really a problem though? Transliterated names are not unique either. Worse-case scenarios, two users will share the same mod path. It won't cause crashes, but it might cause confusion.

SciresM, 2019-01-26:
In addition, this creates a huge amount of cruft throughout codebase -- suddenly *all* of loader needs to run twice, checking both `exefs` and `exefs_<identifier>`, and code needs to be written to only run a second path after account has been initialized and can be connected to. It sounds super, super gross from a code maintenance perspective.

AnalogMan151, 2019-01-27:
I understand. I only suggested it because the current approach is carried over from the 3DS which was a single user system and isn’t quite suited for the multi-user system of the Switch. It just feels odd to have a single-user solution on a multi-user platform. I’m sure it would have been more useful to have approached this from the beginning rather than adapting the current system.

SciresM, 2019-01-27:
I mean I think fundamentally user-profile stuff is just extremely weird to integrate into loader/FS -- it requires talking to account, which is launched fairly late and itself heavily dependent on loader/FS...and the only way the active user gets set is when the profile selector runs, and the whole thing just gets messy.