Atmosphere-NX/Atmosphere #187: creport crashes when generating error reports on SD card since commit a811b447

thedax, 2018-08-11:
I'm on firmware 5.1.0, using tomGER's sdfilesswitch 7.1, and when applications (homebrew or otherwise) crash, creport only generates zero byte log files because it's crashing. The offending function is CodeList::GetFormattedAddressString, but I haven't been able to completely figure out why. When doing anything to the char array `this->address_str_buf`, it will crash creport, generating an additional second blank log file for creport itself.

https://github.com/Atmosphere-NX/Atmosphere/commit/a811b447ce84e09d951766829cc64a339df6f6c7#diff-6b2bbcbfcb49c7d3a4b239a08d237140R131

I've also tried building creport myself against libnx stable 1.3.2 as well as against the latest commit of libnx (fb42ebe53ab5ed6d2e1d3372f5853ef57487029c) to no avail, so I think we can rule out libnx and tomGER's files.

If I replace the function's contents to only return a set string ("Helloworld", or anything else) it seems to work fine, outputting the not-so-useful string as I would expect, and proceed onward to output to the text file(s), as well as the thread dump bin files.

tumGER, 2018-08-11:
Can confirm this happening from multiple independent people using different libnx version so yeah, it's definitely something with creport itself

fennectech, 2018-08-11:
So this is the cause of the 0 byte crash reports?

thedax, 2018-08-11:
Fennectech: to the best of my knowledge, yeah.

closed 2018-08-12 by SciresM

fennectech, 2018-08-12:
Are the error reports being written to the internal flash then?

fennectech, 2018-08-12:
I mean when creport was failing to write them to the sdcard   Does that mean they would hve been written to the internal logs as they normaly would have?  Or  did they wind up in the bit bucket.

SciresM, 2018-08-12:
@fennectech No, when they were failing it was just not writing anything anywhere. They definitely weren't writing to internal flash.

fennectech, 2018-08-12:
Is there any way to reset the switch's logging back to factory ?

SciresM, 2018-08-12:
Not in a way that's undetectable to N.

fennectech, 2018-08-12:
Factory reset?

fennectech, 2018-08-12:
Would initalizing the console without deleting save data get rid of logging?  Or no

SciresM, 2018-08-12:
neither does what it wants you to do, and deleting the erpt savedata is bad too because the savedata tracks (and communicates to N) how many logs it has received historically.

commit 6c8a1a39 (2018-08-12): creport: fix various issues, improve support (closes #187)

-Fixes a NULL deref in thread info parsing
-Heuristically uses [<buildid>] instead of rodata contents if NSO has no
name segment
-Adds full thread/code region dump for all firmwares, instead of just
5.x.