Atmosphere-NX/Atmosphere #188: Games like okami hd and minecraft bedrock crashin an odd way

fennectech, 2018-08-12:
If you try to save in okami HD or use the store in bedrock the console will crash But you can still get the power menu to appear by holding the power button but you cannotinteract with it.   This happens on normal homebrew enabler as well as atmosphere not loading SM seem s to prevent the problem from appearing.

plasmafart, 2018-08-12:
Did you have online disabled?

fennectech, 2018-08-12:
This happens on all homebrew enablers      I do not have online disabled. i however use a DNS server to sinkhole a couple URLs.

aroooo, 2018-08-20:
Crashes on GO VACATION too btw

aroooo, 2018-08-20:
Happen also with and without Online (on ReiNX)

aroooo, 2018-08-27:
I've found a (temporary) fix! Removing fs_mitm sys module should work!

hexkyz, 2018-09-28:
Closing this down as it appears to be related to old bugs in fs_mitm.

closed 2018-09-28 by hexkyz