Atmosphere-NX/Atmosphere #190: make error

itsmine, 2018-08-16:
/usr/gitworkspace/Atmosphere-NX/Atmosphere/stratosphere/loader/source/ldr_content_management.cpp: In static member function 'static Result ContentManagement::MountCodeNspOnSd(u64)':
/usr/gitworkspace/Atmosphere-NX/Atmosphere/stratosphere/loader/source/ldr_content_management.cpp:63:62: error: 'FsFileSystemType_ApplicationPackage' was not declared in this scope
     Result rc = fsOpenFileSystemWithId(&g_CodeFileSystem, 0, FsFileSystemType_ApplicationPackage, path);
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/gitworkspace/Atmosphere-NX/Atmosphere/stratosphere/loader/source/ldr_content_management.cpp:63:62: note: suggested alternative: 'ViServiceType_Application'
     Result rc = fsOpenFileSystemWithId(&g_CodeFileSystem, 0, FsFileSystemType_ApplicationPackage, path);
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                              ViServiceType_Application
/usr/gitworkspace/Atmosphere-NX/Atmosphere/stratosphere/loader/source/ldr_content_management.cpp:63:17: error: 'fsOpenFileSystemWithId' was not declared in this scope
     Result rc = fsOpenFileSystemWithId(&g_CodeFileSystem, 0, FsFileSystemType_ApplicationPackage, path);
                 ^~~~~~~~~~~~~~~~~~~~~~
/usr/gitworkspace/Atmosphere-NX/Atmosphere/stratosphere/loader/source/ldr_content_management.cpp:63:17: note: suggested alternative: 'FsFileSystem'
     Result rc = fsOpenFileSystemWithId(&g_CodeFileSystem, 0, FsFileSystemType_ApplicationPackage, path);
                 ^~~~~~~~~~~~~~~~~~~~~~
                 FsFileSystem
make[5]: *** [ldr_content_management.o] Error 1
make[4]: *** [build] Error 2
make[4]: Leaving directory `/usr/gitworkspace/Atmosphere-NX/Atmosphere/stratosphere/loader'
make[3]: *** [loader] Error 2
make[3]: Leaving directory `/usr/gitworkspace/Atmosphere-NX/Atmosphere/stratosphere'
make[2]: *** [check_stratosphere] Error 2
make[2]: Leaving directory `/usr/gitworkspace/Atmosphere-NX/Atmosphere/fusee/fusee-secondary'
make[1]: *** [fusee-secondary] Error 2
make[1]: Leaving directory `/usr/gitworkspace/Atmosphere-NX/Atmosphere/fusee'
make: *** [fusee] Error 2

GG3Hahn, 2018-08-16:
Those are both declared in libnx's `fs.h` which gets included correctly.
Maybe you're on an older version of libnx?

CaptinNemrac, 2018-08-16:
I’m on the latest version of libnx and I’m getting the same issue. :(

GG3Hahn, 2018-08-18:
For reference:
The enum entry and the function in question were added in [this commit](https://github.com/switchbrew/libnx/commit/d726c7c55c2367b163a6aa7bbc455f1712c09a23) and have been a part of libnx since release v1.3.1

closed 2018-09-22 by SciresM