Atmosphere-NX/Atmosphere #191: Consider switching to another license

filfat, 2018-08-17:
GPL is a cancerous "viral" license that for example would require someone making custom modules to release their source as GPL too. While GPL claims to stand for freedom, this isn't freedom at all.

Please consider switching to like BSD or even better MIT.

SapphireExile, 2018-08-17:
I'm not seeing the issue with GPL.

SciresM, 2018-08-17:
@filfat No, Atmosphere is quite intentionally GPL. I think all custom firmware should be open-source, so that improvements can be shared.

I'm willing to potentially consider making libstratosphere ISC license in the future (which is my non-GPL goto), but Atmosphere itself is definitely staying GPLv2.

closed 2018-08-17 by SciresM

filfat, 2018-08-17:
@SciresM While I do agree that all improvements made to the custom firmware **should** be shared, I really don't agree with the fundamentals that Richard M Stallman and the FSF stands for. Don't get me wrong, I love open source, I just don't like when a since enforces communist-like restrictions.

Releasing libstratosphere under ISC is definitely a step in the right direction however, which I applaud :)

noirscape, 2018-08-19:
> Putting the rest of AMS under a non-GPL license would imply it would be
> okay for a lot more people to use it with little to no credit. (spoiler:
> TX, aka. GW, already do this, infringing the license).
>
> The motive for choosing GPL is "we spent so much time on it, so if anyone
> uses and/or modifies it, please let us view and copy the changes as well".
> I'm not saying BSD/ISC are bad licenses, far from it, but they serve a
> different purpose (where you want your program to be copied/used as many
> time as possible, possibly in corporate code).

@filfat - Even though I myself am critical of people like RMS, the GPL is
objectively a good license, and the protections and demands it makes ensure
that if you make a PUBLIC release of your project that uses the GPL as a
base that nobody can just copy your project wholesale or use it without
also making sure that the code is available under the same conditions.

While for smaller projects, the MIT (or similar OSS licenses such as the
ISC, Apache and so on) license probably suffice, with a project of the size
of Atmosphere, which directly touches on the security of a device, not
making the project GPL (or a similar license) should be considered very
risky for anyone planning to use derivatives, if said derivatives don't
disclose their source code (which the GPL demands).

This comes in addition to the fact that the GPL also makes it very easy for
patches made on forks to be integrated back into upstream (quite simply: if
you release your patch in public, you need to post the source and list the
changes) which should result into less bugs and more features in the
upstream project, which in my eyes anyway should be desirable, especially
for the end-user.

So in my eyes, the GPL (or a license providing similar protections) should
be _required_ for a project such as Atmosphere.