Atmosphere-NX/Atmosphere #192: make nice website

plutooo, 2018-08-17:
make nice website

ThatNerdyPikachu, 2018-08-17:
*make stock bootstrap website

L1lith, 2018-08-18:
I don't know exactly what you would want on the website but if you could describe the layout then I could try designing it

xorinzor, 2018-08-19:
I'd say a website should the lowest priority for now.
Though I'd be happy to help out where needed as the only programming knowledge I have is webdev related (and some java)

Arno500, 2018-08-19:
Cross referencing issue #101

max-arias, 2018-10-17:
Hey, I can make you guys a site if you have a design/etc. Let me know if you're interested, cheers.

jaames, 2018-10-17:
I wouldn't mind helping to design/build the site too, I already had a few ideas based on the splash screen.

noirscape, 2018-11-10:
> a setup video tutorial

I would _strongly_ recommend against this.

Video guides are bad for numerous reasons, not in the least because they cant be updated or amended if need be.

vaibzzz123, 2018-11-25:
I can help develop the site too! We really need to solidify the design and where it will be hosted. The rest will just be a matter of doing it.

micah686, 2018-11-27:
Perhaps we should take advantage of Github Pages for this. Since it will most likely be a static site, we can use Jekyll with github pages. We should also try to figure out what the website will have before we go about making a website

ctsstc, 2018-11-28:
I'm honestly surprised this project survives as open source; As much as it's nice to have a source of truth website I would try to have it as disconnected from this source as much as possible, or just YOLO and ride this train until the shit hits the fan and this has to be added to a private repo somewhere.

I think you only need a static site (github pages should be fine), and if people need a forum they can use one of the top ones out there that are already supporting this project. A link could even be provided from the site to a 3rd party forum.

SeeBeyondMusic, 2018-11-30:
I made a small demo of what a front page could look like. Important links. GitHub, latest build, documentation that would read from github docs/ folder and provide tutorials, and a direct link to reswitched discord for further discussion. 
[atmos.zip](https://github.com/Atmosphere-NX/Atmosphere/files/2632324/atmos.zip)

AkdM, 2018-12-13:
@SeeBeyondMusic simple but very nice. 👍

almmiron, 2019-05-22:
I'm currently learning web design. I'm in to help make a website.

Sky2310, 2019-09-23:
Atmosphere is a no-frills CFW so no need to have a dedicated website like Kosmos which I stay away from as they're just ripping off Atmosphere and packaging it with other homebrew apps. To me, less is more and Github is the perfect place for quick downloads and browsing of updates and changes.

koyuawsmbrtn, 2020-02-05:
I would like to help with making a website for Atmosphère. I could also offer my infrastructure for a small donation in case you need something extremely robust e.g. if thousands of people want to check out the release notes or something.

AkdM, 2020-02-05:
@koyuawsmbrtn I don’t think a dedicated infrastructure is required for that kind of website. A simple static webpage is more than enough, hosted on Github Pages.

Noah670, 2021-02-06:
Did we ever get a website?

ottworks, 2021-06-01:
ok but where is the documentation for getting this on hardware?

SciresM, 2021-06-01:
I don't understand the question -- how you load atmosphere onto the device isn't something ams concerns itself very much with.

if you have an original production unit you can use the RCM bug, @ottworks.

rubenmarcus, 2021-07-06:
@ottworks [here](https://translate.google.com/translate?sl=pt&tl=en&u=https://docs.google.com/document/u/2/d/e/2PACX-1vS60Rb0VffR5s3I8BUbFTogJiTEfqOyii2ZduQ25FUpWbmt-SJ898oX2DI0LWFbbam-rWukSAQZXPqe/pub)

GGLinnk, 2021-07-23:
I'm working on a GitHub page for Atmophere, if that's ok.

I would like to know if SciresM or anyone else is against Reactjs and/or bootstrap?
I'm just asking in case there's some sort of conflict with them...

To see what I've currently done you can check https://gglinnk.github.io/Atmosphere

If you have any kind of suggestions...

hershesz, 2021-07-23:
For a simple page I see no reason why it shouldn’t be static. I think a
static site generator is appropriate for a small size so that JavaScript is
not required to view the page. Also, reactjs has a large footprint and
would increase load times for those on bad connections and break
compatibility with older browsers.

GGLinnk, 2021-07-23:
> For a simple page I see no reason why it shouldn’t be static. I think a static site generator is appropriate for a small size so that JavaScript is not required to view the page. Also, reactjs has a [...]

I would like to write multiple page, one per usage, Home, Download, Instructions, etc.
I can try to use lightweight version of react or use other js framework... I will also use polyfill for old browser compatibility.
I'm using GitHub API to get release so the webite dosn't have to be updated every update.

hershesz, 2021-07-23:
> I would like to write multiple page, one per usage, Home, Download, Instructions, etc.
> 
> I can try to use lightweight version of react or use other js framework... I will also use polyfill for old browser compatibility.
> 
> I'm using GitHub API to get release so the webite dosn't have to be updated every update.
> 
> 
> 
> > For a simple page I see no reason why it shouldn’t be static. I think a static site generator is appropriate for a small size so that JavaScript is not required to view the page. Also, reactjs has a large footprint and would increase load times for those on bad connections and break compatibility with older browsers.

Again I think for a project like this single page works fine. For downloads, link directly to GitHub. If you are REALLY set on react try Preact it's 3kb but again it would not be necessary for a single page. Documentation is already on the wiki so link to that. Polyfills are expensive and the upside for this is, no js needed.

koyuawsmbrtn, 2021-07-23:
> I'm working on a GitHub page for Atmophere, if that's ok.
> 
> I would like to know if SciresM or anyone else is against Reactjs and/or bootstrap?
> I'm just asking in case there's some sort of conflict with them...
> 
> To see what I've currently done you can check [gglinnk.github.io/Atmosphere](https://gglinnk.github.io/Atmosphere)
> 
> If you have any kind of suggestions...
> 
> If I'm not allowed to work on it or to post link, I'm sorry...

Hi, I got a few design suggestions and would really like to help building it.

[image]

hershesz, 2021-07-23:
> Hi, I got a few design suggestions and would really like to help building it.
>
>
>
> [image]

Colors are nice as well with the simplicity but if it's just gonna be the README GitHub can do that just as well. I was thinking maybe have a hero section up above. Larger font and logo in the center along with the boot background. Have the github button front and center with a link to the docs as well

koyuawsmbrtn, 2021-07-23:
Yep, I was thinking the same. I was just improving on something I already had in my hands. We really need things like documentation, current releases, (maybe a blog?). Would be nice to have all that.

GGLinnk, 2021-07-23:
The purpose of this site is to make the download and installation process more user friendly. (And everything about installing sysmodules, etc.) GitHub can be for some users really confusing...

I took all your recommendations, I'll think@koyuawsmbrtn's design is really great.

Maybe I didn't specify it, but it's a github page, not really designed for advanced features, plus the site I gave you is just a test, a prototype, it's far from complete...

TheCatHacker562, 2021-08-13:
here is a website that i made, it is made using html, i expect to be compatible with old web browsers
https://thecathacker562.github.io/atmos/
here source code: https://github.com/TheCatHacker562/atmos

masagrator, 2021-08-13:
> here is a website that i made, it is made using html, i expect to be compatible with old web browsers
> https://thecathacker562.github.io/atmos/
> here source code: https://github.com/TheCatHacker562/atmos

This thread's name is "make **_NICE_** website".

TheCatHacker562, 2021-08-13:
Is in work in progress, I started it like a hour ago.

TheCatHacker562, 2021-08-13:
I need to add alot

TheCatHacker562, 2021-08-13:
update: added another banner, added licences, added discord, added download
web: https://thecathacker562.github.io/atmos/
source code: https://github.com/TheCatHacker562/atmos
if you can't see the changes try deleting history and cookies

SciresM, 2021-12-13:
It's been 3 and a half years. I think we're probably unlikely to get a website as a result of this issue.

If you're particularly interested, reach out to me on discord.

closed 2021-12-13 by SciresM