Atmosphere-NX/Atmosphere #202: fs mitm causes some games to crash upon launch

ThatNerdyPikachu, 2018-09-09:


bakugo, 2018-09-09:
It also causes some older games to crash if you have many files.

For example, simply extracting Splatoon 2's entire romfs and applying it back with layeredfs (no updates or dlc installed) will cause it to crash on startup.

SciresM, 2018-09-10:
Yeah, this is because fs.mitm doesn't use domains at present. It's planned for the future but a lot of work.

duckbill007, 2018-09-20:
Also games can not start if there are 8 or more DLC installed.

ThatNerdyPikachu, 2018-10-01:
we probably could, yeah

fennectech, 2018-10-01:
MITM causes the bedrock Minecraft to crush.

duckbill007, 2018-10-01:
And if this can be configured through text file (I mean tid list) it can be perfect solution for now.

AnalogMan151, 2018-10-01:
Actually, a better work around would be just to not activate LayeredFS if there's no mods present on the SD card. Example tested and working here: https://github.com/Reisyukaku/NX_Sysmodules/blob/3c12add04883b893698cfa04451c834e34bcbe66/fs_mitm/source/fsmitm_service.hpp#L24-L33

fennectech, 2018-10-02:
This is even better!

commit f603dbfc (2018-10-16): fs.mitm: Add domain support (closes #202)
closed 2018-10-16 by SciresM