Atmosphere-NX/Atmosphere #203: warmboot not working

ThatNerdyPikachu, 2018-09-09:
as of https://github.com/Atmosphere-NX/Atmosphere/commit/fa0e9061299eff22095923125ba82ca118e2b5d2, warmboot on 5.1.0 has stopped working for some reason.

SciresM, 2018-09-10:
Are you sure you're setting target firmware appropriately?

In particular, I changed the default firmware from 5.x to 6.x. See [here](https://github.com/Atmosphere-NX/Atmosphere/commit/fa0e9061299eff22095923125ba82ca118e2b5d2#diff-a0b3c028118b80a8a9b1f9bcac5fd95cR38).

This firmware value affects the warmboot security magic number.

ThatNerdyPikachu, 2018-09-14:
hekate master: nope 🤔

ThatNerdyPikachu, 2018-09-15:
this now seems to just be an issue with hekate, is it not setting target fw correctly?
(works fine on fusee^^)

thedax, 2018-09-15:
Also unable to reproduce with hekate 4.0/latest hekate master.

closed 2018-09-15 by SciresM