Atmosphere-NX/Atmosphere #204: Got Error Code 2162-0002 on a XAK10000526349 console which bought in Taiwan

jnxfzhzx, 2018-09-16:
[image]
After I enter Atmospherer CFW and press A to unlock console , the error show up.
I bought this console in 2017.12 and any method to solve the problom?
I also tried to boot in TX OS , it works ,no Error , but no response when I click the Album.

Resaec, 2018-09-16:
Hurry up and shorten your SerialNr. Also remove it from the picture.
Why do you post this online?

noirscape, 2018-09-16:
@Resaec - Switch bans aren't issued with the serial number, they're issued
through console specific data. Merely posting a serial number online isn't
dangerous afaik.

As to why he posted this: Maybe it's to... oh I don't know: submit a bug
report to the issue tracker?

jnxfzhzx, 2018-09-17:
@Resaec OK , I have changed the content as you suggested . But , how to solve this problem?

jnxfzhzx, 2018-09-17:
@noirscape Yes , I want to know how to solve this problem.

hexkyz, 2018-09-17:
This is a general crash error that is often triggered by installed applications and not by the CFW itself. Have you tried a factory reset?

jnxfzhzx, 2018-09-18:
@hexkyz I solve this problem by upgrade tf card when the system promt a request. Thanks all.

closed 2018-09-18 by jnxfzhzx