Atmosphere-NX/Atmosphere #206: Improvement: improve fusée screen printing

Resaec, 2018-09-18:
For 1.0 or later a rework of the current printf/k to a single function to include all message types except fatal_error would be nice.

Supplying a type of message (DEBUG, INFO, MANDATORY, WARNING) to the function would apply colloring as well as prefixes like `[DEBUG] ` to the messages print to screen.

Maybe it could also hide the splash screen in case of type > `MANDATORY` so the user can see that something went wrong as long as there is no SD logging in fusée?

This would also enable users to define the log level in BCT.ini and reduce DEBUG clutter on screen.
Printing to screen is probably quite slow, so this would further improve the boot time.

As printf/k use va_list it would be easiest to make `messageType` the first parameter of this new function.

Please share your thoughts.

hexkyz, 2018-09-20:
I agree, this is exactly something I had envisioned. Now that fusee is functional, I believe it would be preferable for it to be silent (displaying only the logo) unless specifically told to show log messages.

hexkyz, 2018-09-28:
The referenced PR has been merged. I'll work on adding support for choosing logging levels with the BCT.ini file and then this issue can probably be closed.

hexkyz, 2018-09-28:
Implemented logging level configuration from BCT.ini in the latest commit. Closing this now.

closed 2018-09-28 by hexkyz