Atmosphere-NX/Atmosphere #207: Make error when compiling with libnx 1.4.1

Candunc, 2018-09-18:
```
$ make
make -C fusee all
[...]
make -C creport all
make[4]: Entering directory '/home/duncb/NintendoSwitch/Atmosphere/stratosphere/creport'
creport_thread_info.cpp
aarch64-none-elf-g++ -MMD -MP -MF /home/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/build/creport_thread_info.d -g -Wall -O2 -ffunction-sections -march=armv8-a -mtune=cortex-a57 -mtp=soft -fPIE -DDISABLE_IPC -I/home/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/include -I/opt/devkitpro/portlibs/switch/include -I/opt/devkitpro/libnx/include -I/home/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/../libstratosphere/include -I/home/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/build -D__SWITCH__ -fno-rtti -fno-exceptions -std=gnu++17 -c /home/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/source/creport_thread_info.cpp -o creport_thread_info.o
In file included from C:/devkitPro/libnx/include/switch.h:13,
                 from C:/Users/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/source/creport_thread_info.cpp:17:
C:/Users/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/source/creport_thread_info.cpp: In member function 'bool ThreadInfo::ReadFromProcess(Handle, u64, bool)':
C:/Users/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/source/creport_thread_info.cpp:61:43: error: cannot convert 'u8*' {aka 'unsigned char*'} to 'ThreadContext*'
     if (R_FAILED(svcGetDebugThreadContext((u8 *)&this->context, debug_handle, this->thread_id, 0xF))) {
                                           ^~~~~~~~~~~~~~~~~~~~
C:/devkitPro/libnx/include/switch/result.h:12:30: note: in definition of macro 'R_FAILED'
 #define R_FAILED(res)      ((res)!=0)
                              ^~~
In file included from C:/devkitPro/libnx/include/switch.h:23,
                 from C:/Users/duncb/NintendoSwitch/Atmosphere/stratosphere/creport/source/creport_thread_info.cpp:17:
C:/devkitPro/libnx/include/switch/kernel/svc.h:874:48: note:   initializing argument 1 of 'Result svcGetDebugThreadContext(ThreadContext*, Handle, u64, u32)'
 Result svcGetDebugThreadContext(ThreadContext* ctx, Handle debug, u64 threadID, u32 flags);
                                 ~~~~~~~~~~~~~~~^~~
make[5]: *** [/opt/devkitpro/devkitA64/base_rules:14: creport_thread_info.o] Error 1
make[4]: *** [Makefile:118: build] Error 2
make[4]: Leaving directory '/home/duncb/NintendoSwitch/Atmosphere/stratosphere/creport'
make[3]: *** [Makefile:12: creport] Error 2
make[3]: Leaving directory '/home/duncb/NintendoSwitch/Atmosphere/stratosphere'
make[2]: *** [Makefile:123: check_stratosphere] Error 2
make[2]: Leaving directory '/home/duncb/NintendoSwitch/Atmosphere/fusee/fusee-secondary'
make[1]: *** [Makefile:8: fusee-secondary] Error 2
make[1]: Leaving directory '/home/duncb/NintendoSwitch/Atmosphere/fusee'
make: *** [Makefile:5: fusee] Error 2
```

Built on Windows 10, latest version of libnx (1.4.1) and built from repository source.

Resaec, 2018-09-19:
Builds ok on Linux here.
I can reporduce this error in MSYS2.

Candunc, 2018-09-19:
Huh, so it's MSYS2 specific. I'll move to Linux for compilation then

fincs, 2018-09-19:
> Huh, so it's MSYS2 specific. I'll move to Linux for compilation then

Nonsense. This was introduced by a [recent libnx commit](https://github.com/switchbrew/libnx/commit/3ecd841ec2a938edacb887cc7291ca3a0b864d2a) which made it to 1.4.1.

Resaec, 2018-09-19:
d8391078c8828e3cf852742963e4f7f7e0f4ab96 fixed this error on Windows, but on Linux the libnx package from devkitpro seems to be behind.
It doesnt know `ThreadContext` and throws massive undecladed errors
```
In file included from /home/nxdev/nx/repo/Atmosphere/stratosphere/creport/source/creport_thread_info.cpp:20:
/home/nxdev/nx/repo/Atmosphere/stratosphere/creport/source/creport_thread_info.hpp:26:9: error: 'ThreadContext' does not name a type; did you mean 'pthread_once_t'?
         ThreadContext context{};
         ^~~~~~~~~~~~~
         pthread_once_t
```
and so on

fincs, 2018-09-19:
>but on Linux the libnx package from devkitpro seems to be behind.

This is not right. The same repo (dkp-libs) is used to distribute libraries. Have you tried `pacman -Syu`?

Resaec, 2018-09-19:
All up to date.
```
nxdev@nxdev:~/nx/repo/Atmosphere$ git fetch
nxdev@nxdev:~/nx/repo/Atmosphere$ git pull
Already up-to-date.
nxdev@nxdev:~/nx/repo/Atmosphere$ dkp-pacman -Syu
:: Synchronizing package databases...
 dkp-libs is up to date
 dkp-linux is up to date
:: Starting full system upgrade...
 there is nothing to do
nxdev@nxdev:~/nx/repo/Atmosphere$ make -C stratosphere/creport/
make: Entering directory '/home/nxdev/nx/repo/Atmosphere/stratosphere/creport'
creport_thread_info.cpp
aarch64-none-elf-g++ -MMD -MP -MF /home/nxdev/nx/repo/Atmosphere/stratosphere/creport/build/creport_thread_info.d -g -Wall -O2 -ffunction-sections -march=armv8-a -mtune=cortex-a57 -mtp=soft -fPIE -DDISABLE_IPC -I/home/nxdev/nx/repo/Atmosphere/stratosphere/creport/include -I/opt/devkitpro/portlibs/switch/include -I/opt/devkitpro/libnx/include -I/home/nxdev/nx/repo/Atmosphere/stratosphere/creport/../libstratosphere/include -I/home/nxdev/nx/repo/Atmosphere/stratosphere/creport/build -D__SWITCH__ -fno-rtti -fno-exceptions -std=gnu++17 -c /home/nxdev/nx/repo/Atmosphere/stratosphere/creport/source/creport_thread_info.cpp -o creport_thread_info.o
In file included from /home/nxdev/nx/repo/Atmosphere/stratosphere/creport/source/creport_thread_info.cpp:20:
/home/nxdev/nx/repo/Atmosphere/stratosphere/creport/source/creport_thread_info.hpp:26:9: error: 'ThreadContext' does not name a type; did you mean 'pthread_once_t'?
         ThreadContext context{};
         ^~~~~~~~~~~~~
         pthread_once_t
```
Of cause I checked the packet and its 1.4.1-1
```
nxdev@nxdev:~/nx/repo/Atmosphere$ dkp-pacman -Qs libnx
local/libnx 1.4.1-1 (switch-dev)
```

fincs, 2018-09-19:
Sounds to me like the package didn't get installed properly, probably due to someone earlier trying to install from source. Try `sudo dkp-pacman -S libnx --force`

Candunc, 2018-09-19:
Builds successfully as of commit d839107 in MSYS2.

closed 2018-09-19 by Candunc

Resaec, 2018-09-19:
Wow, pacman just lost my trust 👶
Thanks fincs, it's fixed now.

A command to remember

fennectech, 2018-09-19:
As an arch user.  I can agree.  The —force parameter is definitely useful.