Atmosphere-NX/Atmosphere #226: Update https://github.com/Atmosphere-NX/Atmosphere/wiki/release-plans as it is past august

sandhoners123, 2018-10-11:
Update https://github.com/Atmosphere-NX/Atmosphere/wiki/release-plans as it is past August

Resaec, 2018-10-11:
Quoted from the >plan< `"Current best estimate"`
As there is no new estimate, either not available or not needed, there is nothing to change.

Once released one could edit the plan to add the final release date like this `actual relaese on xx`

wpbullet, 2018-10-11:
I think it would be very cool if there was someway to be notified when there is a release like via email or something like that.

fennectech, 2018-10-12:
Soon™

hexkyz, 2018-10-12:
To be honest, 0.7 could've been released nearly a month ago or possibly even on late August which means that estimate was accurate back then.

Motivation decreased significantly during some time due to all the drama surrounding events that were not even directly related to Atmosphere.
Potential end users have also been demonstrating low to no interest in this project since other solutions have been made available.

Additionally, interested users have been able to compile the project themselves for quite a while and contributors still interested in the project often reach us in IRC or Discord to exchange ideas and work on issues.
As a result, honoring a public release has become less and less of a priority while experimental features (either public or private) and fine tuning have been the focus for the latest months.
0.7 will be released this month, we are just no longer trying to rush it.

Of course, I can only speak for myself but I believe @SciresM and @TuxSH have similar opinions.

I'm closing this since it's not a valid issue (the August estimate will be removed after release).

closed 2018-10-12 by hexkyz

noirscape, 2018-10-12:
I would like to note is that the general perception is that Atmosphere is currently not in a usable state (which is false, since as you said, there are currently fully functional builds already being distributed by interested users, and there is only a single issue left before the 0.7 release), which leads to the general perception that Atmosphere can't be used.

This specifically I see as a large number of the reasons as to why end users seem to move other solutions (all of which at this point in time ironically use components and sysmodules _from_ Atmosphere as a base).

Having a stable release of Atmosphere would help to resolve this general perception and would increase your potential amount of currently dwindling end-users.

flareofghast, 2018-10-14:
Yeah, I’ve been hanging out for a stable version.

1232112321, 2018-10-14:
@hexkyz Probably need to disagree. I would prefer major mature interested end users are just not speaking instead of losing their interests. I am hanging my switch untouched still, waiting for the official release of Atmosphere.

hexkyz, 2018-10-14:
I fully agree with all the points being brought up here.
I know that an expressive number of users are mature enough to understand the ups and downs of the project and still support it. I believe, for those, the official release constitutes a nice milestone to be celebrated and my commentary was not directed at them.
For you who keep supporting and helping out the project I have nothing but gratitude. Thank you for being patient and understanding.