Atmosphere-NX/Atmosphere #231: Dark Souls Remastered crash

Lorenzooone, 2018-10-18:
Firmware version: 5.1.0
Language: Italian
Atmosphere version: 0.7
Update: Installed
Connected to the internet: No
Description: When selecting the Female sex, the game hangs up and stops working entirely. Every attempt to go to the home menu fails. It eventually crashes. Discovered by accident.
Repeatable: Yes
Oddity: The crash reports are duplicated, there are two that are full crash reports and two blank ones.

[01539880750_01004ab00a260000.log](https://github.com/Atmosphere-NX/Atmosphere/files/2492827/01539880750_01004ab00a260000.log)

[01539880490_01004ab00a260000.log](https://github.com/Atmosphere-NX/Atmosphere/files/2492828/01539880490_01004ab00a260000.log)

Sonlen1414, 2018-10-18:
Does this also happen on stock to confirm it is an Atmosphere issue and not a corrupted install?

SciresM, 2018-10-18:
Thanks for the report! I'll check this out.

SciresM, 2018-10-18:
It looks like IFile->Read() is failing for some file in Dark Soul's RomFS. I'll try to get in contact with you to give you a debug build of fs.mitm that logs all input/output, so that I can diagnose this better.

Lorenzooone, 2018-10-18:
I think I know what happened and it probably is not Atmosphere's fault. I'll test something out and come back with the results (Will take a while).

Installing it in the SD makes it work...

SciresM, 2018-10-18:
Okay -- please close the issue if you think it's not ams :)

Lorenzooone, 2018-10-18:
Issue was that I used a tool to zero the deleted files in the NAND (before installing the game) and that screwed up with the files.
Luckily I had a backup of before doing that.
Sotty for wasting your time and thanks!

closed 2018-10-18 by Lorenzooone