Atmosphere-NX/Atmosphere #232: [Feature Request] Display Atmosphère next to firmware version

sora10pls, 2018-10-18:
Similarly to many Nintendo 3DS CFWs and ReiNX on Nintendo Switch, they all display the name of the CFW or some sort of identifier letting the user know that they're running CFW when viewing current firmware in System Settings.

Ideally, Atmosphère could display something like `Current version: x.x.x (Atmosphère)`. This could help differentiate if a user booted normally or not, and would definitely come in handy once EmuNAND is fully implemented.

SciresM, 2018-10-18:
This would be good to do.

newget, 2018-10-22:
That string is in settings.msbt.szs.
I don't know how it will be handled in future atmosphere version. But if atmosphere use LayeredFS with whole settings.msbt.szs , it might have issue with horizon korean/chinese localization patch

SciresM, 2018-10-22:
@newget Much better way to do it would be to mitm settings and lie about the firmware version when 809:/file is read.

closed 2018-10-24 by SciresM