Atmosphere-NX/Atmosphere #237: Atmosphere crash on 1.0.0, needs some help guys.

Hayato213, 2018-10-22:
[image]

Hayato213, 2018-10-22:
https://noirscape.github.io/SwitchGuide/sdcard.html

I was following this, crash report wasn't generated. It crash after atmosphere load and go into the home menu. Does it has to do with 1.0.0?

[image]

SciresM, 2018-10-22:
It looks like you're using sys-ftpd.kip. If you remove that from the kips folder, does the crash still occur?

Hayato213, 2018-10-22:
Okay, it stop crashing, thanks SciresM, you helped me again. Last time was on ProjectPokemon Cyber Gadget Save Editor, been years.

SciresM, 2018-10-22:
:) -- sounds like this isn't an atmosphere problem.

closed 2018-10-22 by SciresM

noirscape, 2018-10-22:
FWIW, fixed this in my guide too.

Didn't know sys-ftpd was broken on 1.0.0.