Atmosphere-NX/Atmosphere #239: System Crashes/screen goes black while playing BoTW

Hauteknits, 2018-10-23:
After updating to Atmosphere 7.2, now playing BoTW triggers a crash during certain events, logged so far: *Leaving a shrine *Headshot on Black Moblin. Im not sure if its a crash because they'res no report and the system seems to still be on, but the backlight is off, the screen is black and wont turn back on until the power is held for 10-12 seconds. BoTW was installed via .NSP and the events listed didnt trigger crashes before the update. (Note: SDFiles updater may be to blame, havent tested.)

SciresM, 2018-10-24:
This indicates that Exosphere panicked in some way. Is the crash reproducible? If so and if you can come on ReSwitched's discord I can give you debug builds of Atmosphere that will allow me to identify where the exosphere panic is occurring.

Hauteknits, 2018-10-24:
@SciresM the crash is reproducable, what is the link for the discord

Hauteknits, 2018-10-25:
Also, selecting singleplayer on Mario Kart 8 has triggered a "crash." The system will also not power cycle until it is plugged into a power source. It doesn't seem to matter whether it the included adapter or USB power via my tower.

fennectech, 2018-10-25:
See if holding power button for 15 seconds makes it reboot

SciresM, 2018-10-25:
Talked to @Hauteknits on discord -- it seems like this is a bona fide hardware problem, as his switch crashes whether or not Atmosphere is being used. If it seems like this actually is an ams problem after all, issue will be reopened.

closed 2018-10-25 by SciresM