Atmosphere-NX/Atmosphere #241: [Feature Request] Unlock 5G wifi channel limit of Switch

jonirrings, 2018-10-24:
As some players bought Switch of other region, and the original 5G channels do not compate the local region, 

Thus to change the Router's wifi channel, you have to choose whether Switch to access the 5G wifi or other devices. That is a hard dilemma.
If choose the 2.4G wifi, you have to suffer the too crowded wifi channel, which will cause unstable network even you within 1 meter to Router's wifi.

Is it possible to unlock 5G wifi channel limit of Switch? after the release of ver 1.0?

HylianMedia, 2018-10-24:
Anyone else getting a thomleg50 vibe from reading this, or is just me?

ThatNerdyPikachu, 2018-10-24:
I do too lol

ZachyCatGames, 2018-10-24:
> Anyone else getting a thomleg50 vibe from reading this, or is it just me?

I am as well

jonirrings, 2018-10-25:
what is thomleg50?  I can't read the metaphor.
sorry if I did anything wrong or offensive or broke any rule?
but please reveal it so I could obey it next time.

very sorry again for not knowing the rules

Jon
2018/Otc/25

fennectech, 2018-10-25:
It’s not a brand new account.   I dont think its them.      Thomleg50 is a user who has caused lots of trouble here making issues.      Looks like broken English to me.

jonirrings, 2018-10-25:
I think it conclude to my broken English.
Sorry, I would to study English again. not native English speaker 😭.

I'm not a new comer to Github, and have been developing Web Front-End for 4 years.
but for the hacking area you guys are in, I'm just a game player not a hacker like you you cool guys😜

I will consult my friends to help fix the broken English in this issue.
but if this feature is supposed to be finished by Nintendo, please lock this issue, I will push this feature to Nintendo.

Sorry for bothering.

jonirrings, 2018-10-25:
Surely, the big thing first. I'm waiting for the EmuNAND too😁.
It is not a feature in urgent need and can be avoided by a wired connection.

Thanks for your reply.

fennectech, 2018-10-25:
Your welcome    Happy hacking B)

jonirrings, 2018-10-25:
@fennectech thank you

Let me explain my situation
When I scan the WIFI, I will get a result of 20+ WiFi SSIDs (on 2.4Ghz), and every channel from 1-11 is crowded. (as in China, we got 13 channels, but the most device only support the former 11 channels).

but for the 5Ghz. I get 2 to 5 SSIDs,(the number depending on somewhat magic), and none of the channels is crowded.
Because the coverage of 802.11ac (on 5Ghz) is much smaller than 802.11n (2.4Ghz), my neighbors' WiFi won't interfere mine.

My UBNT AP can change to channel 49, which my HK version Switch supports, but that way, all my other devices willl fail to connect the 5G WiFi.

Again thank you very much.

jonirrings, 2018-10-26:
@james-d-elliott Thank you. 
The lower channels are overlapped. Only one of my devices failed to discover the 5G WiFI AP.

jonirrings, 2018-11-18:
@fennectech I update with a scan results of WiFi on 2.4Ghz. We always got over-crowded 2.4Ghz WiFi channels.

fennectech, 2018-11-19:
According to this a strong router at channel 3 should get a good connection.      You may need to buy a louder router.